ColinPitrat/kalenji-gps-watch-reader #48: Support Navbike 400

Kuiyaire, 2015-04-26:
Salut,
J'ai un compteur GPS Navbike 400 fourni avec un logiciel très limité. J'ai essayé d'utiliser le logiciel Geonaute et il permet de transférer la trace GPS (pas l'alti ni le cardio) sur le PC en se faisant passer pour une OnMove100. J'ai essayé kalenji-gps-watch-reader, mais sans succès.

Sous windows XP SP3 : 
kalenji_reader.exe
error on USB operation code -7 - 0 LIBUSB_ERROR_TIMEOUT

 avec dans le log : 
 => 54 5a 01 01 00 00 ff ff 
 <= 6a 
 <= 6b 
 <= 42 

kalenji_reader.exe -D OnMove100
error on USB operation code -2 - 0 Unknown error

Le device ID est le même que dans le cas de l'OnMove100 : 10c4 ea61

Si tu penses pouvoir supporter ce produit, je peux fournir des dumps USB et des GPX (obtenus avec cette appli https://github.com/bdhurkett/Laser-GPS-Converter-2 qui vient juste d'être adaptée pour le navbike, mais qui ne fait pas le transfert USB, seulement l'export en GPX)

ColinPitrat, 2015-04-26:
Bonjour,

je suis intéressé pour le supporter !
N'y a-t-il pas un logiciel fourni avec la montre ? Si oui, est-il possible de fournir un dump réalisé avec ainsi qu'un dump réalisé avec Geonaute et le gpx associé, afin que je puisse essayer de comprendre le protocole ?

Si nécessaire, la page https://github.com/ColinPitrat/kalenji-gps-watch-reader/wiki/Produce-dumps explique comment produire des dumps.

Cordialement,
Colin Pitrat

Kuiyaire, 2015-04-27:
Il y a bien un logiciel appelé Ultrasport GPS fourni avec le compteur , qui ne permet pas d'exporter de fichiers mais simplement de consulter les traces et exporter/importer les réglages du compteur.

D'autres logiciels du même genre qui correspondent au même compteur sous une marque différente permettent d'exporter en TCX ou directement sur Strava (Outbond GPS Sport Watch, Soleus Draft GPS, Laser Navbike GPS).

J'ai fait quelques dumps dispos ici : http://1drv.ms/1GoAciY
- compteur vide : import par le logiciel Ultrasport puis Geonaute (avec débranchement/branchement USB)
-compteur avec une trace de quelques secondes : import par Utrasport, puis Geonaute et export de la trace GPS par Geonaute.

Le GPS n'a pas eu le temps de prendre l'altitude apparemment.
Si besoin je pourrai faire d'autres dumps plus complet avec cardio, alti, pause, lap, etc...

ColinPitrat, 2015-08-29:
Bonjour,

désolé d'avoir laissé traîner ce ticket aussi longtemps !

Après avoir regardé les dumps en détail, j'ai l'impression que c'est effectivement très proche de l'OnMove 100. La seule différence que je puisse voir est dans la réponse à l'une des commandes d'initialisation qui est légèrement différente à ce que j'attends. 

Voici une version modifié du programme qui devrait corriger ce problème. Il faut d'abord essayer avec le device OnMove100:
**kalenji_reader.exe -D OnMove100**

La sortie devrait être beaucoup plus verbeuse que précédemment. En cas d'erreur, cela permettra de comprendre ce qui ne se passe pas bien.

Si le transfert fonctionne, tant mieux ! Le Navbike 400 est alors semblable au OnMove100 et peut être utilisé de la même manière.

Si ce n'est pas le cas, il faut essayer:
**kalenji_reader.exe -D Navbike400**

J'ai en effet écrit une version plus proche du dump pour ce qui concerne les phases d'initialisation, même si les différences que j'ai relevé ne devrait pas avoir d'impact.

Merci de me tenir au courant du résultat !

Cordialement,
Colin Pitrat

Kuiyaire, 2015-08-31:
Bonjour,

Merci d'avoir essayé d'adapter ton prog.

Avec le nouveau kalenji_reader.exe j'ai les mêmes messages d'erreur suivant :

kalenji_reader.exe -D OnMove100 -v
src/main.cc:329: Configuration parsed
src/main.cc:333: Create output directory 'c:\Prog\kalenji'
src/main.cc:349: Source is USB
src/main.cc:353: With transaction logger
src/main.cc:410: Registering device
src/main.cc:417: Attaching source to device
src/main.cc:420: Initializing device
src/device/OnMove100.cc:36: OnMove100::init() - init devic
src/device/OnMove100.cc:43: OnMove100::init() - step 1
error on USB operation code -2 - 0 Unknown error

kalenji_reader.exe -D Navbike400 -v
src/main.cc:329: Configuration parsed
src/main.cc:333: Create output directory 'c:\Prog\kalenji'
src/main.cc:349: Source is USB
src/main.cc:353: With transaction logger
src/main.cc:410: Registering device
src/main.cc:417: Attaching source to device
src/main.cc:420: Initializing device
src/device/Navbike400.cc:36: Navbike400::init() - init device
src/device/Navbike400.cc:43: Navbike400::init() - step 1 (frame 2)
src/device/Navbike400.cc:46: Navbike400::init() - step 2 (frame 4)
src/device/Navbike400.cc:49: Navbike400::init() - step 3 (frame 6)
src/device/Navbike400.cc:52: Navbike400::init() - step 4 (frame 9)
src/device/Navbike400.cc:55: Navbike400::init() - step 5 (frame 12)
src/device/Navbike400.cc:58: Navbike400::init() - step 6 (frame 15)
src/device/Navbike400.cc:77: Navbike400::init() - device init attempt 1
src/device/Navbike400.cc:84: Navbike400::init() - step 13 (retry device init)
src/device/Navbike400.cc:77: Navbike400::init() - device init attempt 2
src/device/Navbike400.cc:84: Navbike400::init() - step 13 (retry device init)
src/device/Navbike400.cc:77: Navbike400::init() - device init attempt 3
src/device/Navbike400.cc:84: Navbike400::init() - step 13 (retry device init)
src/device/Navbike400.cc:77: Navbike400::init() - device init attempt 4
src/device/Navbike400.cc:84: Navbike400::init() - step 13 (retry device init)
src/device/Navbike400.cc:77: Navbike400::init() - device init attempt 5
src/device/Navbike400.cc:84: Navbike400::init() - step 13 (retry device init)
src/device/Navbike400.cc:77: Navbike400::init() - device init attempt 6
src/device/Navbike400.cc:84: Navbike400::init() - step 13 (retry device init)
src/device/Navbike400.cc:77: Navbike400::init() - device init attempt 7
src/device/Navbike400.cc:84: Navbike400::init() - step 13 (retry device init)
src/device/Navbike400.cc:77: Navbike400::init() - device init attempt 8
src/device/Navbike400.cc:84: Navbike400::init() - step 13 (retry device init)
src/device/Navbike400.cc:77: Navbike400::init() - device init attempt 9
src/device/Navbike400.cc:84: Navbike400::init() - step 13 (retry device init)
src/device/Navbike400.cc:77: Navbike400::init() - device init attempt 10
src/device/Navbike400.cc:84: Navbike400::init() - step 13 (retry device init)
src/main.cc:422: Device initialized
src/main.cc:424: Get sessions list
src/device/Navbike400.cc:123: Navbike400: Get sessions list !
error on USB operation code -7 - 0 LIBUSB_ERROR_TIMEOUT

avec dans le log :
 => ee ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 => ee ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 => ee ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 => ee ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 => ee ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 => ee ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 => ee ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 => ee ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 => ee ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 => ee ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 => 08 08 aa aa 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

ColinPitrat, 2015-08-31:
Super, c'est plutôt bon signe pour la version Navbike 400 !

Grace à la sortie et aux logs, j'ai changé deux petites choses qui, je l'espère, devraient aider:
https://drive.google.com/file/d/0BweL-sTrQlo_aFRXWFdtTm5IWFU/view?usp=sharing

Pouvez-vous réessayer avec:
kalenji_reader.exe -D Navbike400 -v

Merci !

Kuiyaire, 2015-09-01:
Merci du travail, ça va dans le bon sens :)
```
kalenji_reader.exe -D Navbike400 -v
src/main.cc:329: Configuration parsed
src/main.cc:333: Create output directory 'c:\Prog\kalenji'
src/main.cc:349: Source is USB
src/main.cc:353: With transaction logger
src/main.cc:410: Registering device
src/main.cc:417: Attaching source to device
src/main.cc:420: Initializing device
src/device/Navbike400.cc:36: Navbike400::init() - init device
src/device/Navbike400.cc:43: Navbike400::init() - step 0 (frame 18 (empty))
src/device/Navbike400.cc:46: Navbike400::init() - step 1 (frame 2)
src/device/Navbike400.cc:49: Navbike400::init() - step 2 (frame 4)
src/device/Navbike400.cc:52: Navbike400::init() - step 3 (frame 6)
src/device/Navbike400.cc:55: Navbike400::init() - step 4 (frame 9)
src/device/Navbike400.cc:58: Navbike400::init() - step 5 (frame 12)
src/device/Navbike400.cc:61: Navbike400::init() - step 6 (frame 15)
src/device/Navbike400.cc:80: Navbike400::init() - device init attempt 1
src/device/Navbike400.cc:80: Navbike400::init() - device init attempt 2
src/device/Navbike400.cc:80: Navbike400::init() - device init attempt 3
src/device/Navbike400.cc:80: Navbike400::init() - device init attempt 4
src/device/Navbike400.cc:87: Navbike400::init() - step 13 (retry device init)
src/device/Navbike400.cc:80: Navbike400::init() - device init attempt 5
src/device/Navbike400.cc:80: Navbike400::init() - device init attempt 6
src/device/Navbike400.cc:80: Navbike400::init() - device init attempt 7
src/device/Navbike400.cc:80: Navbike400::init() - device init attempt 8
src/device/Navbike400.cc:80: Navbike400::init() - device init attempt 9
src/device/Navbike400.cc:80: Navbike400::init() - device init attempt 10
src/main.cc:422: Device initialized
src/main.cc:424: Get sessions list
src/device/Navbike400.cc:126: Navbike400: Get sessions list !
src/device/Navbike400.cc:129: Navbike400: Read sessions list !
src/device/Navbike400.cc:155: Session from: 2015-8-27 14:3:20
src/device/Navbike400.cc:155: Session from: 2015-8-30 14:39:6
src/device/Navbike400.cc:155: Session from: 2015-8-30 17:13:48
src/device/Navbike400.cc:155: Session from: 2015-8-30 18:42:6
src/device/Navbike400.cc:175: Navbike400: Read session !
src/device/Navbike400.cc:227: Navbike400: Finished !
src/main.cc:430: Filter out sessions
Sessions available for import:
    0 - 2015-08-27 12:03:20     0 laps          0 km              0s
    1 - 2015-08-30 12:39:06     0 laps          0 km              0s
    2 - 2015-08-30 15:13:48     0 laps          0 km              0s
    3 - 2015-08-30 16:42:06     0 laps          0 km              0s
List of sessions to import (space separated - 'all' to import everything - 'new'
 to import only new sessions):
all
src/main.cc:433: Get sessions details
src/main.cc:436: Release device
src/device/Navbike400.cc:97: Navbike400::release() - step 1 (frame 32)
src/main.cc:441: Release datasource
Filter does not exist:
Creating c:\Prog\kalenji/20150827_140320.gpx
Creating c:\Prog\kalenji/20150827_140320.kml
Creating c:\Prog\kalenji/20150827_140320.html
```

Il n'y a qu'un GPX créé qui correspond vaguement à la session 0, mais trop court et avec parfois des coordonnées correctes, mais pas les bons temps. Il n'y a pas non plus l'altitude et les pulsations.
Si je lance la commande plusieurs fois, le GPX et le fichier log créés ne sont pas à chaque fois de la même taille.

Si besoin, voici le GPX créé par le programme, le log, ainsi que les 4 GPX des 4 sessions obtenus par le logiciel fourni avec le compteur +export GPX avec "Laser GPS converter" : http://1drv.ms/1UjxEj1

ColinPitrat, 2015-09-01:
C'est très intéressant !
Les  0 laps / 0 km / 0s sont "normaux" dans l'aperçu des sessions, ils ne sont pas (encore ?) supportés (à voir si j'arrive à trouver où ils sont). Est-ce que les dates et heures sont bien correctes ? Est-ce que la montre supporte les temps intermédiaires (laps) ?

En comparant 0.gpx et 20150827_140320.gpx, on voit que le second contient les 876 premiers points du premier (sauf quelques points en plus, 2 ou 3) ainsi que pas mal de points supplémentaires avec latitude et longitude à 0. En regardant les logs, on voit effectivement qu'il y a des points sans informations correspondant probablement à des moments où la montre perd le GPS.

Par contre effectivement les dates et heures n'ont rien à voir et ne correspondent pas avec celles de l'aperçu. La façon dont c'est géré est différent de l'OnMove. Visiblement il y a un point chaque seconde, est-ce que cela se règle ? Si oui serait il possible de faire une session (même très courte) avec un réglage différent ?

Le plus embêtant est le fait que le téléchargement s'interrompe avant la fin. Cela vient du fait que des données parasites sont insérées au milieu d'un point (a priori pas toujours le même ?). Pour mieux comprendre, trois choses peuvent m'aider:
 - avoir plus d'exemples de logs de transferts avec kalenji_reader (échouant plus ou moins loin, le gpx n'est pas nécessaire)
 - reessayer le transfert avec une version encore modifiée du programme (cf ci-dessous). Les données ne seront probablement pas meilleures mais les logs devraient être plus complets
 - avoir les dumps du transfert fait avec "Laser GPS Converter" (on peut garder cette option pour le cas où les deux tentatives précédentes n'aboutissent pas dans la mesure ou ça génère plus de travail ...)

Voici donc encore une nouvelle version dans laquelle j'ai essayé d'améliorer quelques infos (heure et altitude des points) et de tolérer trop ou trop peu de données dans un point pour obtenir plus de logs.
https://drive.google.com/file/d/0BweL-sTrQlo_Nm92LVZ3VFc1SVU/view?usp=sharing

Merci.
Colin

ColinPitrat, 2015-09-05:
Bonjour, 

avez-vous eu l'occasion de tester la dernière version ?

Colin

ColinPitrat, 2016-02-13:
Bonjour,

avez-vous eu l'occasion de tester ?

Colin

Kuiyaire, 2016-02-29:
Salut,
Vraiment désolé de répondre si tardivement, j'avais oublié :(

Je ne me souviens plus trop des manips que j'avais faites avant, mais voici ce que j'obtiens avec la version que tu as mises en lien (une session sur le compteur, de 2h) : 
`src/main.cc:329: Configuration parsed
src/main.cc:333: Create output directory 'c:\Prog\kalenji'
src/main.cc:364: Source is USB
src/main.cc:368: With transaction logger
src/main.cc:425: Registering device
src/main.cc:432: Attaching source to device
src/main.cc:435: Initializing device
src/device/Navbike400.cc:35: Navbike400::init() - init device
src/device/Navbike400.cc:42: Navbike400::init() - step 0 (frame 18 (empty))
src/device/Navbike400.cc:45: Navbike400::init() - step 1 (frame 2)
src/device/Navbike400.cc:48: Navbike400::init() - step 2 (frame 4)
src/device/Navbike400.cc:51: Navbike400::init() - step 3 (frame 6)
src/device/Navbike400.cc:54: Navbike400::init() - step 4 (frame 9)
src/device/Navbike400.cc:57: Navbike400::init() - step 5 (frame 12)
src/device/Navbike400.cc:60: Navbike400::init() - step 6 (frame 15)
src/device/Navbike400.cc:79: Navbike400::init() - device init attempt 1
src/main.cc:437: Device initialized
src/main.cc:439: Get sessions list
src/device/Navbike400.cc:125: Navbike400: Get sessions list !
src/device/Navbike400.cc:128: Navbike400: Read sessions list !
src/device/Navbike400.cc:154: Session from: 2016-2-26 14:59:12
src/device/Navbike400.cc:192: Navbike400: Read session !
src/device/Navbike400.cc:228: Point not ending with 0xfa - skipping
src/device/Navbike400.cc:228: Point not ending with 0xfa - skipping
src/device/Navbike400.cc:228: Point not ending with 0xfa - skipping
src/device/Navbike400.cc:270: Navbike400: Finished !
src/main.cc:445: Filter out sessions
Sessions available for import:
    0 - 2016-02-26 13:59:12     0 laps          0 km              0s
List of sessions to import (space separated - 'all' to import everything - 'new'
 to import only new sessions):
all
src/main.cc:448: Get sessions details
src/main.cc:451: Release device
src/device/Navbike400.cc:96: Navbike400::release() - step 1 (frame 32)
src/main.cc:456: Release datasource
Filter does not exist:
Creating c:\Prog\kalenji/20160226_145912.gpx
Creating c:\Prog\kalenji/20160226_145912.kml
Creating c:\Prog\kalenji/20160226_145912.html
`
Et voila le log : https://onedrive.live.com/redir?resid=D61C3BE5F0093448!678&authkey=!AB8WUQnmbUeFJ2c&ithint=folder%2clog

N'hésite pas a demander des compléments, je reçois par mail les notifs :)

ColinPitrat, 2016-05-07:
Bonjour,

désolé d'avoir mis un peu de temps à répondre.
Pour ce qui est de l'altitude, j'ai corrigé ça (c'est dans git) mais je n'ai pas encore fait de release avec.
Je veux bien un exemple avec les laps afin de les supporter. Comme ça une fois que ce sera fait, je pourrai fermer ce ticket ! :-)

Colin

ColinPitrat, 2019-05-17:
On va considérer ça comme implémenté. On verra si quelqu'un sent le besoin d'avoir les laps un jour.

closed 2019-05-17 by ColinPitrat