ColinPitrat/kalenji-gps-watch-reader #50: Core dump with OnMove100 when several tracks stay in the watch

yasbow, 2015-12-06:
Kalenji GPS watch reader works fine with OnMove100 watch, but I have noticed that program crashes when several old tracks are already in the watch and when we want to extract last record.
With OnMove 100, we cannot remove track after exporting it to computer (although this is possible with original windows software, but it's another issue), so if there is still 3 or 4 tracks in the watch, the software crashes and there is no way to get tracks with kalenji reader (but tracks are still available with windows software).

What steps will reproduce the problem ?
1. Connect the watch On Move 100
2. Launch kalendji-reader
3. Core dump

What is the expected output ? What do you see instead ?
There is no output data file.

What version of the product are you using ? On what operating system?
Product version 4.9.3
OS : Linux 3.13.0-71-generic #114-Ubuntu SMP Tue Dec 1 02:34:22 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

I have attached core generated from a debug version of kalenji_reader and associated console output 
[crash.zip](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/53205/crash.zip)
(and all crash information generated by ubuntu). If you need some more information or test, I will help you with pleasure.

ColinPitrat, 2016-02-13:
Hello,

sorry for the delay in answering.

One thing that would help me much is to have the logs of the upload. They are activated by default and can be found in a subdirectory log of your output directory.

With those logs I should be able to reproduce the issue and understand more clearly why this happens (probably an assumption I make about the way the data is transmitted that is not always true).

Regards,
Colin

yasbow, 2016-02-14:
Hello,

You can find joined to this mail the corresponding logs. I am not really 
sure they are the good ones, but the date matches with the reported bug 
(3 files because I made several tries to extract data).
Hope this can help.
If the logs are not enough to fix the problem, I can try to reproduce it 
with new records.

Regards
Phil

ColinPitrat, 2016-02-15:
Hello,

it looks like attaching files to the mail didn't work (as it's going through github). Or maybe you forgot it ?
What you can do is to send me the files by mail directly. My e-mail adress is simply my first name dot my last name on gmail dot com.

Regards,
Colin

ColinPitrat, 2016-02-16:
Hello,

Did you update the firmware of your watch ?

The log files seems to be completely different from what I have for OnMove100 examples.

In the initialisation sequence, the answer is not the same I had in the examples I had. It may be an information about differences in the formatting of the data, it's hard to say.

However, what follows (the real data about the session) is completely different. I fixed the core, but even with that, I think either you will not manage to import anything or what you will import will be rubish.

Regards,
Colin

yasbow, 2016-02-18:
Hello,

I don't think I have updated my firmware (I don't really remember, I have this watch since 2012).
I can send you a log file generated when export works fine to compare with the previous one, if you want.
I will try the new version with several records, but I understand that it won't be enough to get correct data.

A workaround is to empty the watch after each export, but this must be done by running the original windows software. Is it possible to add this feature to your software ?  When I do it with geonaute software I have this strange message in a popup window (looks like sequence in your log files) :
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FD 00 00 FF FD

Thanks again for your work.
Phil

yasbow, 2016-04-30:
Hello,
I have tested the fix with my watch in which several tracks have been recorded. No more crash, I have the following output :

ERROR: Unknown session 1
ERROR: Unknown session 3
ERROR: Unknown session 5
Sessions available for import:
    0 - 2228-01-31 08:41:51     0 laps          0 km              0s
    2 - 2018-01-31 17:52:14     0 laps          0 km              0s
    4 - 2238-02-01 01:26:28     0 laps          0 km              0s
    6 - 2000-03-31 02:00:36     0 laps          0 km              0s
    7 - 2075-12-10 15:49:06     0 laps          0 km              0s
List of sessions to import (space separated - 'all' to import everything - 'new' to import only new sessions): 
all
  Applying filter FixElevation
    Fixed elevation of 66 points.
  Applying filter ComputeSessionStats
Ascent is 9 and descent is 4
Creating 22280131_094151.gpx
Creating 22280131_094151.html
  Applying filter FixElevation
    Fixed elevation of 67 points.
  Applying filter ComputeSessionStats
Ascent is 7 and descent is 25
Creating 20180131_185214.gpx
Creating 20180131_185214.html
  Applying filter FixElevation
    Fixed elevation of 67 points.
  Applying filter ComputeSessionStats
Ascent is 17 and descent is 20
Creating 22380201_022628.gpx
Creating 22380201_022628.html
  Applying filter FixElevation
    Fixed elevation of 66 points.
  Applying filter ComputeSessionStats
Ascent is 4 and descent is 7
Creating 20000331_040036.gpx
Creating 20000331_040036.html
  Applying filter FixElevation
    Fixed elevation of 6931 points.
  Applying filter ComputeSessionStats
Ascent is 2319 and descent is 2318
Creating 20751210_164906.gpx
Creating 20751210_164906.html

Only one track is correct, the last one. The other tracks have only some old points. As you can see, all dates are wrong, but it's the only problem I noticed. But in fact, it's a good result, because, I have connected the watch on Windows using the original software, and it found just one track, the last one, as kalenji-reader did. So the patch is a good workaround for the issue.

If you need the logs for this extract, eventually to fix date issue, I can send it to you.
Thank you.
Phil

ColinPitrat, 2016-04-30:
The logs would be nice to try to fix the date, but also to see if I can find a way to detect 'wrong' sessions and simply ignore them. Thanks for your help making kalenji_reader better.

yasbow, 2016-05-01:
These are the logs of the last export with wrong and good sessions.
If you need something else, you're welcome.
[kalenji_reader_20160429_123244.log.zip](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/244027/kalenji_reader_20160429_123244.log.zip)

ColinPitrat, 2016-05-02:
I understood better the issue. The headers for all sessions but the last are incomplete. I don't know what this means: the detail of the sessions (the points) are still in the watch but the information about the sessions (e.g date & time) is missing.

This is easy to detect and I modified the code to handle this case. In your example, it now downloads only the last session with the correct date.

If you confirm this fixes your problem, I'll close the issue.

yasbow, 2016-05-02:
It works as you predicted : sessions from 0 to 6 are seen as wrong sessions, and the last is correctly read with date and time. The output is the following :
ERROR: Unknown session 0
ERROR: Unknown session 1
ERROR: Unknown session 2
ERROR: Unknown session 3
ERROR: Unknown session 4
ERROR: Unknown session 5
ERROR: Unknown session 6
Sessions available for import:
    7 - 2016-04-22 08:09:12     0 laps          0 km              0s
List of sessions to import (space separated - 'all' to import everything - 'new' to import only new sessions): 
all
  Applying filter FixElevation

and after I received for some points (not all) an error from Google API : OVER_QUERY_LIMIT. I don't know why, but it's not the problem.

As this watch is bit limited, when the storage capacity is reached, it automatically erases old sessions to get free space for current points. In fact, I suppose that it just erases older headers and overwrite points on previous sessions. In this example, the last session lasts more than 6 hours and 40 minutes, it's probably near watch storage limit ...

Thank you again for the fix, you can close this issue.

ColinPitrat, 2016-05-02:
Perfect !
I also removed the "ERROR" message unless in verbose mode.
For the Google API error, it's probably because you reached the max number of queries per day. If I remember well, it corresponds to roughly 7 hours with 1 point per second so if you did multiple import in less than 24 hours it can trigger this.

closed 2016-05-02 by ColinPitrat