ColinPitrat/kalenji-gps-watch-reader #51: GPS On Move 200

DOCPROF, 2016-02-11:
Hello,

I have a watch GPS on Move 200 and i work with Linux Ubuntu.
I use your program with this syntax :
./kalenji_reader -D OnMove710 -p /media/user/ONMOVE-200
There is no error but under /tmp/kalenji_import I have no file.
can you consider this device ? I can help you if yoy want.

DOCPROF, 2016-02-11:
Désolé je viens de voir que tu es français, donc pour préciser mes tests, j'au aussi utilisé le device OnMove100 et à ce moment j'ai une erreur de segment.
Si tu peux m'aider.
Merci.

ColinPitrat, 2016-02-11:
Bonjour,

Le segfault est une mauvaise chose de toute façon, il faudrait attacher les logs (avec l'option -v) pour voir si ça aide. Ou ajouter un core avec un build en debug:
make clean && make debug
Pour que le core soit généré, il faudra peut être faire:
ulimit -c unlimited

Un run avec l'option -v pour le OnMove710 peut être intéressant aussi:
./kalenji_reader -v -D OnMove710 -p /media/user/ONMOVE-200

Je ne connais pas le OnMove 200, il faut voir comment il fonctionne. Si c'est comme le OnMove 710, il est vu comme un USB mass storage (une clef USB par exemple) et il faut décoder des fichiers. Ce semble être le cas puisque tu as un path /media/user/ONMOVE-200. Si c'est comme le OnMove 100 il faut échanger directement par USB. 

Evidemment, la version USB mass storage est plus simple puisqu'il suffit de me copier les fichiers et de me fournir les .gpx correspondant pour que je puisse regarder à quoi ressemble le protocole (ou si tu veux regarder toi même).

Cordialement,
Colin

DOCPROF, 2016-02-11:
Bonjour,
Je te confirme, il est vu comme un stockage de masse.
J'accède bien au répertoire DATA où je trouve des fichiers .OMD et .OMH
C'est ces fichiers que tu désires ?
En ajoutant l'option -v j'ai ce message :

src/main.cc:329: Configuration parsed
src/main.cc:333: Create output directory '/tmp/kalenji_import/'
src/main.cc:410: Registering device
src/main.cc:417: Attaching source to device
src/main.cc:420: Initializing device
src/main.cc:422: Device initialized
src/main.cc:424: Get sessions list
src/main.cc:430: Filter out sessions
src/main.cc:433: Get sessions details
src/main.cc:436: Release device

merci de ton aide :+1:

ColinPitrat, 2016-02-11:
Alors ce n'est pas le même format que le OnMove710 (en tout cas les extensions ne sont pas les mêmes).
Pour pouvoir supporter cette montre, il faudrait m'envoyer un exemple de fichier OMD / OMH et le GPX correspondant produit par le logiciel officiel.

DOCPROF, 2016-02-11:
Voici les fichiers, je te joins le GPX qui j'ai récupéré après avoir utilisé l'application de geonaute.
Pour info j'ai utilisé aussi l'outil : [https://github.com/rboninsegna/OnMove200](https://github.com/rboninsegna/OnMove200)
Cet outil lit les fichiers .OMH avec mes fichiers joints j'ai ce résultat : 
jean-fixe OnMove200-0.9 # ./om200-omh ../GPS/ACT_0001.OMH
Data file ../GPS/ACT_0001.OMH:
 File number: 2
 Activity started on: 2016-2-10 19:7
 Duration:   0h 16m 46s
 Distance:   15.79 km (15789 m)
 Speed:      56.50 km/h
 Top Speed:  86.81 km/h
 Energy:     1359 kcal

Mais je pense que c'est le .OMD le plus intéressant.

[GPS.zip](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/126913/GPS.zip)

ColinPitrat, 2016-02-12:
Sais-tu le vendor ID & device ID de la montre (visible avec un lsusb) ?

DOCPROF, 2016-02-12:
Je suis en déplacement, je te réponds lundi.

ColinPitrat, 2016-02-12:
J'ai commité une version basique du support pour cet montre.
Ca ne devrait pas trop mal marcher, à tester.

DOCPROF, 2016-02-15:
Bonjour,

Tout d'abord en faisant un lsusb, j'ai ceci :+1: Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 003: ID 046d:c03e Logitech, Inc. Premium Optical Wheel Mouse (M-BT58)
Bus 007 Device 002: ID 04b3:3025 IBM Corp. NetVista Full Width Keyboard
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 283e:0009  
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

Autre chose : J'ai oublié de te préciser que les parcours sont stockés dans un répertoire DATA.
Tous les parcours sont stockés sous forme de 2 fichiers 
ACT_0000.OMH toujours de 60 octets avec les informations mentionnées plus haut
et ACT_0000.OMD qui contient le parcours.
La séquence 0000 s'incrémente à chaque nouveau parcours mémorisé.

Pour répondra à tes questions : 
-> Je n'ai pas activé le cardio (nécessite une option en plus).
-> L'altitude doit être présente car quand j'utilise Windows :-1:  je vois l'altitude (cf copie jointe) à moins que cela utilise les données de google.
[image]

Je teste ton commit et je te fais un retour.

DOCPROF, 2016-02-15:
Pas mal :clap: 

J'ai du simplement préciser le répertoire DATA sur ma ligne de commande : 
./kalenji_reader -D OnMove200 -p /media/user/ONMOVE-200/DATA

Du coup cela génère bien 2 fichiers : html et GPX.
Au niveau de HTML j'ai bien le parcours mais les courbes ne s'affiche pas :cry: 
Au niveau du GPX, je l'ai importé sur le site de decathlon et fait une comparaison avec le même parcours directement récupéré via le logiciel sous windows. J'ai des différences, voici quelques exemples (en orange le fichier GPX, en bleu avec le logiciel sous windows) : 
[image]

Autre exemple sur la vitesse : 
[image]
Si tu veux plus d'infos n'hésites pas.

Et bravo déjà pour ce premier résultat.

ColinPitrat, 2016-02-16:
OK, les différences sont quand même assez minimes a priori :-)

Pour le lsusb, je suppose que c'est le device 283e:0009. Est-ce qu'il disparaît bien quand tu retires la montre ?

Pour l'altitude, c'est pour ça que je demandais si l'information était disponible d'une manière ou d'une autre sur la montre. Car elle ne semble pas présente dans les fichiers, ni dans l'OMH pour un min/max ou pour le dénivelé positif/négatif, ni dans l'OMD pour les informations sur chaque point. Il est possible que sur Geonaute, l'information soit récupérée sur internet (google map ou autre). 

Pour ce qui est des différences, il faudrait comparer les fichiers gpx pour comprendre d'où elles viennent. Avec l'exemple que tu avais fournit, je n'avais pas de différence significative dans la sortie (juste l'arrondi qui était différent pour la latitude et la longitude, mais c'était le 7e ou 8e chiffre après la virgule !). Quels filtres as-tu activé ? (tu peux voir la liste de ceux qui sont appliqué au moment de l'import).

Je pense que les seuls filtres utiles seraient ComputeInstantSpeed et FixElevation

Pour simplifier la ligne de commande, tu peux tout mettre dans le fichier de configuration (sauf device pour qu'on puisse vérifier que le vendorId / deviceId permet la detection automatique). Cf https://github.com/ColinPitrat/kalenji-gps-watch-reader/wiki/Configuration-File. Je te propose aussi de désactiver les extensions GPX si tu n'as ni temps intermediaires ('laps') ni cardio.

Cela donne le contenu suivant:
source=Path
path=/media/user/ONMOVE-200/DATA
filters=FixElevation,ComputeInstantSpeed
gpx_extensions=none

ColinPitrat, 2016-02-16:
I changed the deviceId / vendorId, so now, the device should be automatically discovered.

DOCPROF, 2016-02-16:
Ok tout fonctionne correctement.
Cependant quand je précise le filtre FixElevation cela fonctionne mais j'ai cette erreur : 
`Error: received an error from Google API: "REQUEST_DENIED".
Error when parsing result of http://maps.googleapis.com/maps/api/elevation/json?sensor=true&locations=47.2594,2.21693|47.2594,2.21693|47.2594,2.21692|47.2594,2.21692|47.2594,2.21692|47.2594,2.21693|47.2594,2.21693|3.1e-05,0
`
Pour les différences j'ai aussi cherché de mon côté et je n'ai rien trouvé (tu es d'ailleurs plus précis). Je te propose de faire un autre parcours et de t'envoyer les résultats.

Pour répondre à ta question, oui le bon device et celui-ci : Bus 004 Device 004: ID 283e:0009 

Merci pour ton travail, il ne te reste plus qu'à modifier ton wiki pour prendre en compte cette nouvelle montre :smiley:

ColinPitrat, 2016-02-16:
Effectivement il y a un point bizarre dans cette requête ! (coordonnées
3.1e-0.5,0 )
C'est avec un autre fichier que celui que tu m'as fourni (de mon côté je
n'ai pas ce problème) ?

ColinPitrat, 2016-02-16:
Je peux mettre ton nom sur la liste des personnes ayant aidé pour le
support d'une montre.
Sous quel nom veux-tu apparaître ?

DOCPROF, 2016-02-16:
Comme évoqué, je vais faire une sortie cette après-midi et vérifier que tout est OK. Ne cherche pas pour le problème c'est peut-être lors d'une sortie que j'ai fait des tests avec la montre (par défaut ton programme récupère toutes les sorties).

Pour le reste je te réponds directement sur ton email.

ColinPitrat, 2016-02-16:
C'est quand même gênant si l'altitude pour certains points ne peut pas être récupérée. C'est possible que ça vienne de point où le signal GPS était perdu. Si tu retrouves l'exemple avec les logs correspondant (dans le sous-repertoire logs à l'endroit où sont générés les fichiers, un fichier de log par execution du programme) ça m'intéresse !

DOCPROF, 2016-02-17:
Nouveau test effectué -> Tout est OK
Correction du warning qui dépend de l'architecture, le champs tm_wday n'étant pas toujours présent -> OK
Merci encore pour ce travail :+1:

closed 2016-02-17 by DOCPROF

Uncledens, 2020-12-14:
Bonjour ColinPitrat, 

J'avais vu ta contribution, alors que je cherchais un programme qui me convertirait mes fichiers OMD de mamontre ONMOVE500 car suite à une mise à jour Windows, il devenait impossible d'envoyer mes fichiers vers l'application Decathlon en ligne. Ma nouvelle montre devenait donc inutile. Mais je ne suis pas linuxien, et j'étais bloqué (sans doute pour pas grand chose) devant mon Raspberry... Étant plus à l'aise avec Python, j'ai repris quelques pistes de ton programme pour retranscrire ce que j'avais compris de la structure du binaire, et ai produit ce week-end un petit programme que j'ai mis sur mon site Web http://uncledens.chez-alice.fr/python/montregps.htm (faut absolument que je refasse ma charte graphique, les couleurs Uncle Ben's sont trop flashies!). 

Voila je voulais me manifester pour te remercier, car ton programme m'a permis de voir qu'il était possible d'attaquer un binaire (je n'avais jamais essayé) et de sortir les info contenues dedans. Il me reste à automatiser la génération de graphiques et la génération de fichiers GPX en limitant le nombre de points (720 points pour une heure, c'est trop). Il faudra que je commente le code et le diffuse également librement sur Github (je ne suis pas vraiment utilisateur). 

Merci encore.
Denis