ColinPitrat/kalenji-gps-watch-reader #52: Compiled kalenji_reader on OSX but I'm having issues with OnMove 200 device...

fpb, 2016-08-14:
Hi!

First of all, thank you for your wonderful piece of software. I managed to compile it under OSX 10.11.6 but I am facing some problems. My device is a OnMove 200...

First, some info on what I changed to compile it:

I downloaded, compiled and installed from source both libusb-1.0 and libcurl. Both got installed in /usr/local. libxml2 was already installed under /usr (I suspect it comes with the os).

I didn't have pkg-config so I manually edited the Makefile. I will try again with pkg-config installed to minimize the changes for OSX.

I managed to import a workout (in fact it was a car ride to test the GPS accuracy). It imported it ok but with an extra point at the end near Lat=0, Long=0. However, subsequent imports generate a segmentation fault after releasing the device. 

Next I tried to compile it with debug (needed to remove -Og option) and I got the following output:

./test/unit/unit_tester --gtest_shuffle
Note: Randomizing tests' orders with a seed of 28474 .
[==========] Running  tests from  test cases.
[----------] Global test environment set-up.
[----------]  tests from EmptyLapsTest
[ RUN      ] EmptyLapsTest.NullDurationLapsRemoved
[       OK ] EmptyLapsTest.NullDurationLapsRemoved ( ms)
[ RUN      ] EmptyLapsTest.NullDistanceLapsRemoved
[       OK ] EmptyLapsTest.NullDistanceLapsRemoved ( ms)
[ RUN      ] EmptyLapsTest.NoLapRemovedIfNoEmptyLap
[       OK ] EmptyLapsTest.NoLapRemovedIfNoEmptyLap ( ms)
[ RUN      ] EmptyLapsTest.SameStartAndEndPointsLapsRemoved
[       OK ] EmptyLapsTest.SameStartAndEndPointsLapsRemoved ( ms)
[----------]  tests from EmptyLapsTest ( ms total)

[----------]  tests from ReducePointsTest
[ RUN      ] ReducePointsTest.LessPointsThanMax
    Reduced session from 10 to 2 points (min angle = pi/63).
test/unit/filter/TestReducePoints.cc:: Failure
      Expected: 10u
      Which is: 
To be equal to: aSession.getPoints().size()
      Which is: 
[  FAILED  ] ReducePointsTest.LessPointsThanMax ( ms)
[ RUN      ] ReducePointsTest.MorePointsThanMaxRemovesStraightLines
    Reduced session from 250 to 3 points (min angle = pi/4).
test/unit/filter/TestReducePoints.cc:: Failure
      Expected: 6u
      Which is: 
To be equal to: aSession.getPoints().size()
      Which is: 
[  FAILED  ] ReducePointsTest.MorePointsThanMaxRemovesStraightLines ( ms)
[ RUN      ] ReducePointsTest.MorePointsThanMax
    Reduced session from 60 to 2 points (min angle = pi/63).
[       OK ] ReducePointsTest.MorePointsThanMax ( ms)
[----------]  tests from ReducePointsTest ( ms total)

[----------]  tests from UtilsTest
[ RUN      ] UtilsTest.durationAsStringConvertsDurationToStringWithMsLowerThan100
test/unit/Utils.cc:: Failure
      Expected: "9d 3h49m2s027"
To be equal to: testString.c_str()
      Which is: ""
[  FAILED  ] UtilsTest.durationAsStringConvertsDurationToStringWithMsLowerThan100 ( ms)
[ RUN      ] UtilsTest.ThrowStreamThrows
[       OK ] UtilsTest.ThrowStreamThrows ( ms)
[ RUN      ] UtilsTest.FormatterStreamAndImplicitStr
[       OK ] UtilsTest.FormatterStreamAndImplicitStr ( ms)
[ RUN      ] UtilsTest.testDirDoesntExistDoNotCreate
Error: /tmp/UtilsTest doesn't exist and I couldn't create it
[       OK ] UtilsTest.testDirDoesntExistDoNotCreate ( ms)
[ RUN      ] UtilsTest.splitStringFourTokens
[       OK ] UtilsTest.splitStringFourTokens ( ms)
[ RUN      ] UtilsTest.ThrowStreamExceptionContainsMessage
[       OK ] UtilsTest.ThrowStreamExceptionContainsMessage ( ms)
[ RUN      ] UtilsTest.testDirOnRegularFile
Error: /tmp/UtilsTest is not a directory
[       OK ] UtilsTest.testDirOnRegularFile ( ms)
[ RUN      ] UtilsTest.splitStringOneToken
[       OK ] UtilsTest.splitStringOneToken ( ms)
[ RUN      ] UtilsTest.str_to_int_42
test/unit/Utils.cc:: Failure
      Expected: 42u
      Which is: 
To be equal to: str_to_int("42")
      Which is: 
[  FAILED  ] UtilsTest.str_to_int_42 ( ms)
[ RUN      ] UtilsTest.distanceEarthParisLondon
[       OK ] UtilsTest.distanceEarthParisLondon ( ms)
[ RUN      ] UtilsTest.trimStringTerminalTabsAndSpaces
[       OK ] UtilsTest.trimStringTerminalTabsAndSpaces ( ms)
[ RUN      ] UtilsTest.str_to_int_trailing_chars
test/unit/Utils.cc:: Failure
      Expected: 42u
      Which is: 
To be equal to: str_to_int("42azerty")
      Which is: 
[  FAILED  ] UtilsTest.str_to_int_trailing_chars ( ms)
[ RUN      ] UtilsTest.durationAsStringConvertsDurationToString
test/unit/Utils.cc:: Failure
      Expected: "10d 7h42m20s"
To be equal to: testString.c_str()
      Which is: ""
[  FAILED  ] UtilsTest.durationAsStringConvertsDurationToString ( ms)
[ RUN      ] UtilsTest.testDirWithParentDirARegularFile
Error: An element in /tmp/UtilsTest/UtilsTest is not a directory
[       OK ] UtilsTest.testDirWithParentDirARegularFile ( ms)
[ RUN      ] UtilsTest.FormatterStreamAndStr
[       OK ] UtilsTest.FormatterStreamAndStr ( ms)
[ RUN      ] UtilsTest.testDirDoesntExistCreate
[       OK ] UtilsTest.testDirDoesntExistCreate ( ms)
[ RUN      ] UtilsTest.trimStringInitialTabsAndSpaces
[       OK ] UtilsTest.trimStringInitialTabsAndSpaces ( ms)
[ RUN      ] UtilsTest.testDirNoPermissions
test/unit/Utils.cc:: Failure
      Expected: -1
      Which is: 
To be equal to: testDir("/root/UtilsTest", true)
      Which is: 
[  FAILED  ] UtilsTest.testDirNoPermissions ( ms)
[ RUN      ] UtilsTest.durationAsStringConvertsDurationToStringWithMs
test/unit/Utils.cc:: Failure
      Expected: "7d 6h34m55s327"
To be equal to: testString.c_str()
      Which is: ""
[  FAILED  ] UtilsTest.durationAsStringConvertsDurationToStringWithMs ( ms)
[ RUN      ] UtilsTest.str_to_int_0
[       OK ] UtilsTest.str_to_int_0 ( ms)
[ RUN      ] UtilsTest.str_to_int_only_chars
[       OK ] UtilsTest.str_to_int_only_chars ( ms)
[----------]  tests from UtilsTest ( ms total)

[----------]  tests from ComputeSessionStatsTest
[ RUN      ] ComputeSessionStatsTest.AscentDescentComputationWhenGoingDown
Ascent is 0 and descent is 100
[       OK ] ComputeSessionStatsTest.AscentDescentComputationWhenGoingDown ( ms)
[ RUN      ] ComputeSessionStatsTest.MaxAndAvgSpeedComputation
Ascent is 22200 and descent is 22856
[       OK ] ComputeSessionStatsTest.MaxAndAvgSpeedComputation ( ms)
[ RUN      ] ComputeSessionStatsTest.DurationComputation
Ascent is 0 and descent is 100
[       OK ] ComputeSessionStatsTest.DurationComputation ( ms)
[ RUN      ] ComputeSessionStatsTest.DistanceComputation
Ascent is 2794 and descent is 0
[       OK ] ComputeSessionStatsTest.DistanceComputation ( ms)
[ RUN      ] ComputeSessionStatsTest.AscentDescentComputationWhenClimbing
Ascent is 100 and descent is 0
[       OK ] ComputeSessionStatsTest.AscentDescentComputationWhenClimbing ( ms)
[----------]  tests from ComputeSessionStatsTest ( ms total)

[----------]  tests from NullHeartrateTest
[ RUN      ] NullHeartrateTest.FirstPointNotFixed
[       OK ] NullHeartrateTest.FirstPointNotFixed ( ms)
[ RUN      ] NullHeartrateTest.BetweenTwoPointsWithDifferentValues
[       OK ] NullHeartrateTest.BetweenTwoPointsWithDifferentValues ( ms)
[ RUN      ] NullHeartrateTest.BetweenTwoPointsWithSameValue
[       OK ] NullHeartrateTest.BetweenTwoPointsWithSameValue ( ms)
[----------]  tests from NullHeartrateTest ( ms total)

[----------] Global test environment tear-down
[==========]  tests from  test cases ran. ( ms total)
[  PASSED  ]  tests.
[  FAILED  ]  tests, listed below:
[  FAILED  ] ReducePointsTest.LessPointsThanMax
[  FAILED  ] ReducePointsTest.MorePointsThanMaxRemovesStraightLines
[  FAILED  ] UtilsTest.durationAsStringConvertsDurationToStringWithMsLowerThan100
[  FAILED  ] UtilsTest.str_to_int_42
[  FAILED  ] UtilsTest.str_to_int_trailing_chars
[  FAILED  ] UtilsTest.durationAsStringConvertsDurationToString
[  FAILED  ] UtilsTest.testDirNoPermissions
[  FAILED  ] UtilsTest.durationAsStringConvertsDurationToStringWithMs

 8 FAILED TESTS
make: *** [unit_test] Error 1

Also, when I run the debug build I immediately get an error (the verbose flag was on):

kalenji_reader(24912,0x7fff76b3b000) malloc: *** error for object 0x102ec7a20: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6

The release version outputs:

 session 0 already imported
Segmentation fault: 11

Thanks!

fpb, 2016-08-15:
Update: curiously, If I managed to get it working by simply setting the path in the config file. I was setting both the path and the device...

Are the problems with the unit tests problematic?

ColinPitrat, 2016-08-18:
Thanks for the feedback. I'm happy that this builds on OS X. If this works after setting the path in the config file, this is a good start. Does the result of the import seem correct ?

The segmentation fault and the failing unit tests highlights portability bugs that it would be interesting to understand and fix. From the output however, I don't get what the issue(s) could be.

fpb, 2016-08-19:
Update: I have both release and debug versions compiled. Debug version gives me a malloc error:

```
kalenji_reader_dbg(2943,0x7fff73f85000) malloc: *** error for object 0x1077efa20: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6
```

Running the release version works ok:
```
 session 0 already imported
 session 1 already imported
 session 2 already imported
```
I am also attaching the differences I have from you code to compile on OSX El Capitan, in case you wish to integrate OSX support. I am willing to make any additional tests you need.

```
git diff bd670a2d5ae01f6d907381964edcf0773c28fd93
diff --git a/Makefile b/Makefile
index 9670f56..9babcfd 100644
--- a/Makefile
+++ b/Makefile
@@ -1,12 +1,12 @@
 TARGET=kalenji_reader
-INCPATH=-Isrc $(shell pkg-config --cflags libusb-1.0) $(shell pkg-config --cflags libxml-2.0) $(shell pkg-config --cflags libcurl)
-LIBS=$(shell pkg-config --libs libusb-1.0) $(shell pkg-config --libs libxml-2.0) $(shell pkg-config --libs libcurl)
+INCPATH=-Isrc $(shell pkg-config --cflags libusb-1.0) $(shell pkg-config --cflags libcurl) -I/usr/include/libxml2
+LIBS=$(shell pkg-config --libs libusb-1.0) $(shell pkg-config --libs libxml2) $(shell pkg-config --libs libcurl) -lxml2
 WINOBJECTS=$(shell find src -name \*.cc | sed 's/.cc/.os/')
 OBJECTS=$(shell find src -name \*.cc | sed 's/.cc/.o/')
 HEADERS=$(shell find src -name \*.h)
-CFLAGS=-Wall -Wextra -Wno-unused-parameter -std=c++11
+CFLAGS=-Wall -Wextra -Wno-unused-parameter -std=c++11 -stdlib=libstdc++
 ADD_CFLAGS=-O2
-DEBUG_ADD_CFLAGS=-D DEBUG=1 -D _GLIBCXX_DEBUG -Og -g -coverage -pthread
+DEBUG_ADD_CFLAGS=-D DEBUG=1 -D _GLIBCXX_DEBUG -g -coverage -pthread
 MINGW_PATH=/usr/i686-w64-mingw32
 WININCPATH=-I$(MINGW_PATH)/include/libusb-1.0/ -I$(MINGW_PATH)/include/libxml2/
 WINLIBS=$(MINGW_PATH)/lib/libusb-1.0.dll.a $(MINGW_PATH)/lib/libxml2.dll.a $(MINGW_PATH)/lib/libcurl.dll.a
@@ -78,7 +78,6 @@ windows: $(TARGET).exe win
 
 check_deps:
        @pkg-config --libs libusb-1.0 >/dev/null 2>&1 || (echo "Error: missing dependency libusb-1.0. Try installing libusb development package (e.g: libusb libusb-1 libusb-1.0.0-dev ...)" && false)
-       @pkg-config --libs libxml-2.0 >/dev/null 2>&1 || (echo "Error: missing dependency libxml2. Try installing libxml2 development package (e.g: libxml2 libxml2-dev ...)" && false)
        @pkg-config --libs libcurl >/dev/null 2>&1 || (echo "Error: missing dependency libcurl. Try installing libcurl development package (e.g: libcurl-dev libcurl4-gnutls-dev ...)" && false)
 
 $(OBJECTS): %.o:%.cc $(HEADERS)
diff --git a/src/device/OnMove200.h b/src/device/OnMove200.h
index d700573..f03e276 100644
--- a/src/device/OnMove200.h
+++ b/src/device/OnMove200.h
@@ -1,12 +1,12 @@
-#ifndef _DEVICE_GEONAUTE710_HPP_
-#define _DEVICE_GEONAUTE710_HPP_
+#ifndef _DEVICE_GEONAUTE200_HPP_
+#define _DEVICE_GEONAUTE200_HPP_
 
 #include "Device.h"
 
 namespace device
 {
        /**
-         The interface for OnMove 710 GPS watch. 
+         The interface for OnMove 200 GPS watch. 
         */
        class OnMove200 : public Device
        {
```

The imported files seem to be ok when compared to the ones imported the OnConnect utility.

Cheers!

ColinPitrat, 2016-08-19:
Thanks. Having to remove `pkg-config` for libxml2 to replace it by a hardcoded value is a bit sad. Do you know why pkg-config is not finding info for libxml2 ? Was it properly installed ?

I guess the `-Og` doesn't work because the version of gcc is older than 4.8 ?

For the `-stdlib=libstdc++`, the problem is that it's not supported on the version of gcc I have on my laptop (I didn't check elsewhere). I need to find a way to make it conditional on building for OS X. What's the impact of not adding it ?

fpb, 2016-08-20:
Hi!

> Thanks. Having to remove pkg-config for libxml2 to replace it by a hardcoded value is a bit sad. Do you know why pkg-config is not finding info for libxml2 ? Was it properly installed ?

I managed to get pkg-config to work with the system installed libxml2. Since I couldn't write to /usr/lib/pkconfig I added a /usr/local/lib/pkgconfig/libxml-2.0.pc file with the following contents:

```
prefix=/usr
exec_prefix=${prefix}
libdir=${prefix}/lib
includedir=${prefix}/include

Name: libXML
Description: libXML library version2.
Version: 2.9.2
Libs: -L${libdir} -lxml2
Cflags: -I${includedir}/libxml2
```
However, this is still a hack and not very convenient. But I really don't know how pkg-config can handle standard libraries on OSX.
Also, to compile it on a plain vanilla OSX system, we need to instal pkg-config, libcurl and libusb-1.0 beforehand.

> I guess the `-Og` doesn't work because the version of gcc is older than 4.8 ?

OSX doesn't use gcc anymore. It uses the APPLE LLVM version 7.0 compiler. I tried setting CXX to llvml-g++ but no difference...

Compiler complains with:

```
error: invalid integral value 'g' in '-Og'
```

> For the `-stdlib=libstdc++`, the problem is that it's not supported on the version of gcc I have on my laptop (I didn't check elsewhere). I need to find a way to make it conditional on building for OS X. What's the impact of not adding it ?

If I don't add it I get this:
```
c++ -Wall -Wextra -Wno-unused-parameter -std=c++11 -O2 -c -Isrc -I/usr/local/include/libusb-1.0 -I/usr/include/libxml2 -I/usr/local/include -o src/device/CWKalenji500SD.o src/device/CWKalenji500SD.cc
In file included from src/device/CWKalenji500SD.cc:1:
In file included from src/device/CWKalenji500SD.h:4:
In file included from src/device/Device.h:6:
In file included from src/device/../bom/Session.h:10:
In file included from src/device/../bom/../bom/Lap.h:4:
In file included from src/device/../bom/Point.h:9:
src/device/../bom/../Utils.h:8:10: fatal error: 'tr1/memory' file not found
#include <tr1/memory>
         ^
1 error generated.
make: *** [src/device/CWKalenji500SD.o] Error 1
```

Thanks

ColinPitrat, 2016-08-20:
`tr1` is a leftover of pre-c++11 code. This is not needed anymore, it should work without `-stdlib=libstdc++` once removed.

`-Og` is a nice feature of gcc optimizing developer's experience but doesn't bring much for kalenji_reader, I'll replace it by `-O0` as a simple fix to this issue :-)

Finally, the only thing that remains is:

> Also, to compile it on a plain vanilla OSX system, we need to instal pkg-config, libcurl and libusb-1.0 beforehand.

I'll need to document this. If you have more info (e.g: links), I'm interested.

ColinPitrat, 2016-08-20:
If I'm not mistaken, with my last commit and your hack to make pkg-config work, you should be able to build from the git repository without any modification to it.

fpb, 2016-08-20:
Hi!

I just cloned your repo and managed to build it on a "virgin" OS X El Capitan.

Here are the instructions (using Homebrew):

- **Install xcode command line tools**

`$ xcode-select --install`

- **Install home-brew**

`$ ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)" < /dev/null 2> /dev/null`

- **Install pkg-config via homebrew:**

`$ brew install pkg-config`

- **Install libusb via homebrew:**

`$ brew install libusb`

- **Setup pkg-config to find system libraries:**

`export PKG_CONFIG_PATH=/usr/local/Library/ENV/pkgconfig/10.10`

Note: replace 10.10 with whatever is the latest OS X version appearing listed under pkgconfig

- **make kalenji-gps-watch-reader**

`$ git clone https://github.com/ColinPitrat/kalenji-gps-watch-reader.git`
`$ cd cd kalenji-gps-watch-reader`
`$ make`

ColinPitrat, 2016-08-22:
Thanks. I just added your instructions on a new wiki page:
https://github.com/ColinPitrat/kalenji-gps-watch-reader/wiki/Install-and-use-(Mac-OS-X)

I also added your name in the list of contributors in the README.md, tell me if this is not OK.

I think this issue can be closed, although it would be nice to understand where the core in debug and the failing unit tests come from.

closed 2016-08-22 by ColinPitrat

fpb, 2016-08-22:
I will resume this after getting back from some short vacations.