ColinPitrat/kalenji-gps-watch-reader #53: Problem with OnMove 200

FredCoadou, 2016-08-25:
Hello Colin, I used to use the GPS reader with One Move 100.
I've just bought a OneMove 200.
Your reader recognized One Move 200 as an USB device.
It's really faster than for One Move 100.
The reader did his job.
Sometime the file.gpx is not good. Sometime it is good.
htlm files are ok.
Via onConnect everything is ok.
I send you one go and return files, htlm files. One gpx files is ok and the other is wrong.
I call them .txt

Can you help me, please ?

[20160825_193300.html.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/437826/20160825_193300.html.txt)
[20160825_193300.gpx.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/437827/20160825_193300.gpx.txt)
[20160825_184600.html.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/437829/20160825_184600.html.txt)
[20160825_184600.gpx.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/437828/20160825_184600.gpx.txt)

FredCoadou, 2016-09-04:
I send you OneMove200 files in Data for a wrong gpx file.
The wrong think is the last position, if i cut it the gpx is ok.
[20160902_195100.gpx.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/454029/20160902_195100.gpx.txt)
[20160902_195100.html.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/454030/20160902_195100.html.txt)
[ACT_0000.OMH.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/454031/ACT_0000.OMH.txt)
[ACT_0000.OMD.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/454032/ACT_0000.OMD.txt)

ColinPitrat, 2017-01-19:
Hi,

sorry for the late answer. If I understand correctly, it's only one point that is wrong at the end ?
I think I've already seen a bug mentionning this. Looking at the code, I found the issue. This should be fixed now (assuming you use code from git repo).

I close the issue, please re-open it if it is not fixed.

closed 2017-01-19 by ColinPitrat