ColinPitrat/kalenji-gps-watch-reader #54: ONMove 220: not known USB device

jellium, 2016-09-03:
Hi,
Thanks for the initiative and work!
I just bought a Geonaute ONMove 220 today and went for my first run with it. 

I quickly found your available work here, cloned the repo and compiled everything smoothly. Unfortunately, `./kalenji_reader -v` outputs:
`
src/main.cc:292: Configuration parsed
src/main.cc:296: Create output directory '/tmp/kalenji_import'
src/main.cc:312: Source is USB
src/main.cc:316: With transaction logger
src/main.cc:336: Auto-detecting device
No known USB device found.
`
and therefore I cannot dump the GPX data.

I can see on the repository's main page that the ONmove 220 is indeed not known to be supported, but is there anything I can dig into or pieces of information I can bring you in order to make it work?

Thanks!

jellium, 2016-09-06:
Hi again,

If it helps, I have [uploaded](https://github.com/jellium/onmove220) the `OMD` and `OMH` files created along with my run, and the corresponding `GPX` file exported from the `decathloncoach.com` application.

Also, the USB device ID is `283e:001e`.

Please let me know if you may need anything else or if there is something I can do to help you with.

Merci !

ColinPitrat, 2016-09-06:
One thing you can try is to use it with the OnMove200 driver. As it also uses OMD and OMH files, it may work. For this use `./kalenji_reader -v -D OnMove200`

If it doesn't work, then I'll have a look at the file you uploaded to see if it can easily be supported.

jellium, 2016-09-07:
Here is the output from `./kalenji_reader -v -p /media/jellium/ONMOVE-220/ -D OnMove200 -d /home/jellium/kalenji_reader`:

```
src/main.cc:292: Configuration parsed
src/main.cc:296: Create output directory '/home/jelle/Geonaute/'
src/main.cc:373: Registering device
src/main.cc:380: Attaching source to device
src/main.cc:383: Initializing device
src/main.cc:385: Device initialized
src/main.cc:387: Get sessions list
OnMove200: Get sessions list !
src/main.cc:393: Filter out sessions
src/main.cc:396: Get sessions details
src/main.cc:399: Release device
OnMove200: Release (nothing to do)
```

There is some progress but not file is created. Note that the watch does contain a session ready to be uploaded.

Thanks for your help.

ColinPitrat, 2017-01-20:
Sorry to come back so late on this. I tested with your files and it seems to work in my case.
What did you put in your configuration file ?

In my case I have the following output (running without -v):
```
OnMove200: Get sessions list !
Decode summary of session ACT_0000
Retrieve session ACT_0000
OnMove200: Release (nothing to do)
Creating /tmp/20160906_072900.gpx
```

And the resulting GPX file seems correct. The configuration file I use is:
```
directory=/tmp
source=Path
import=all
log_transactions=no
filters=none
outputs=GPX
```

bouzin, 2018-06-10:
Hello Colin,
I have a new problem. My watch in sot detected anymore. I red this issue and try this :
tom@tom-Vostro-1700:~/kalenji-gps-watch-reader$ sudo ./kalenji_reader -v -D Keymaze
And it's done that : 
src/main.cc:292: Configuration parsed
src/main.cc:296: Create output directory '/tmp/kalenji_import'
src/main.cc:373: Registering device
src/main.cc:380: Attaching source to device
src/main.cc:383: Initializing device
Erreur de segmentation (core dumped)

May you please help an other time ?
Thank you very much
Bouzin

ColinPitrat, 2018-08-02:
One thing you can try is to build it in debug and run it under gdb:

$ make clean
$ make debug
$ sudo gdb ./kalenji_reader
(gdb) r -v -D Keymaze

Then when it coredumps, do:
(gdb) bt

And send me the result.

bouzin, 2018-08-04:
Hi Colin,
Thank you for your answer.I did a new installation after having remove cleanly the old one. I installed git properly and now it works very well.
Thank a lot
Thomas Ferry

 Envoyé le : Jeudi 2 août 2018 10h42

hd1984, 2019-03-23:
Thank you very much ColinPitrat for this tool

The solution for the problem is to add /DATA in the path option.

Replace USERNAME with your username:
./kalenji_reader -v -p /media/USERNAME/ONMOVE-220/DATA/ -D OnMove200 -d /home/USERNAME/Desktop/

closed 2020-06-15 by ColinPitrat