ColinPitrat/kalenji-gps-watch-reader #59: kalenji_reader ne s'execute pas

FredCoadou, 2016-12-19:
Bonjour, j'ai téléchargé votre programme, compilé par "make", normalement celà devrait fonctionner, mais j'ai ce message d'erreur, votre programme fonctionne sur un autre ordinateur, ai je fais une erreur ?
Le fichier kalenji_reader existe pourtant, et il est bien exécutable...
Je suis sous ubuntu 16.4. J'ai réglé les différents problème de dépendances au fur et à mesure...
Merci.

xxxxx@Grobill:~$ cd kalenji-gps-watch-reader
xxxxxx@Grobill:~/kalenji-gps-watch-reader$ ./kalenji_reader
Error: path '' does not exist (check option -p <path> on command line or line path=<path> in configuration file).
error std::exception
xxxxxxx@Grobill:~/kalenji-gps-watch-reader$

ColinPitrat, 2017-01-19:
Le programme se plaint que le path (l'option -p en ligne de command ou path= dans le fichier de configuration) n'a pas été fourni.

Cette information est nécessaire pour les montres utilisant l'USB mass storage (c'est à dire fonctionnant comme une clef USB). C'est le cas, entre autre, de l'OnMove200.

Je pense qu'une commande du genre (en supposant que la montre est montée sur /mnt/removable):
$ ./kalenji_reader -p /mnt/removable
devrait fonctionner.

FredCoadou, 2017-01-20:
Bonjour, merci de l'aide. Voici mon fichier de configuration : 
```
#.kalenji_readerrc
device=OnMove200
import=new
source=Path
path=/media/coadou/ONEMOVE-200/DATA	
filters=FixElevation,ComputeInstantSpeed
directory=/home/coadou/GPS
```

Voici les différentes erreurs que celà me retourne, suivant mes demandes...
```
coadou@Grobill:~/kalenji-gps-watch-reader$ ./kalenji_reader -p /media/coadou/ONMOVE-200/DATA -D OneMove200
Error trying to register device OneMove200: Unknown device
error std::exception
coadou@Grobill:~/kalenji-gps-watch-reader$ ./kalenji_reader
Error: path '' does not exist (check option -p <path> on command line or line path=<path> in configuration file).
error std::exception
coadou@Grobill:~/kalenji-gps-watch-reader$ ./kalenji_reader -p /media/coadou/ONMOVE-200/DATA -D OneMove200
Error trying to register device OneMove200: Unknown device
error std::exception
coadou@Grobill:~/kalenji-gps-watch-reader$ ./kalenji_reader -p /media/coadou/ONMOVE-200/DATA 
Can't autodetermine device with a source different from USB. Specify your device using -D option or device= in your configuration file.
coadou@Grobill:~/kalenji-gps-watch-reader$ 
```
 Si tu as une idée...

ColinPitrat, 2017-01-20:
Je pense que le fichier de configuration n'est pas trouvé, d'où l'erreur de la deuxième commande.
Pour info, que donne la commande en passant le fichier de configuration avec l'option -c ?
`./kalenji_reader -c <config>` 

L'erreur sur la première et la troisième commande viennent du nom du device: c'est OnMove200 et non OneMove200.

FredCoadou, 2017-01-20:
```
coadou@Grobill:~/kalenji-gps-watch-reader$ ./kalenji_reader -c <config>
bash: erreur de syntaxe près du symbole inattendu « newline »
coadou@Grobill:~/kalenji-gps-watch-reader$ 
```

ColinPitrat, 2017-01-20:
Je voulais dire de mettre le chemin vers le fichier de config à la place de "<config>", par exemple:
./kalenji_reader ~/.kalenji_readerrc

FredCoadou, 2017-01-20:
```
[coadou@Grobill:~/kalenji-gps-watch-reader$ ./kalenji_reader -c/home/coadou/.kalenjirc
Error: path '/media/coadou/ONEMOVE-200/DATA' does not exist (check option -p <path> on command line or line path=<path> in configuration file).
error std::exception
coadou@Grobill:~/kalenji-gps-watch-reader$ ]

```

FredCoadou, 2017-01-20:
Pardon, je viens de corriger le seconde erreur sur le fichier de configuration :*
```
coadou@Grobill:~/kalenji-gps-watch-reader$ ./kalenji_reader -c/home/coadou/.kalenji_readerrc
Error: path '' does not exist (check option -p <path> on command line or line path=<path> in configuration file).
error std::exception
coadou@Grobill:~/kalenji-gps-watch-reader$ 
```

ColinPitrat, 2017-01-20:
J'avais oublié de demander d'ajouter l'option -v pour avoir plus d'info mais je ne pense pas que ça aide dans ce cas.
La ligne "path= " dans le fichier de configuration contient peut être un caractère binaire non visible ? Cela devrait permettre de vérifier:
`grep path /home/coadou/.kalenji_readerrc | hexdump -C`

Une autre possibilité est d'essayer de la ré-écrire complètement à la main.

Qu'en est-il de la commande avec `-D OnMove200` ?

FredCoadou, 2017-01-20:
Ho je suis navrée, je viens de voir ce qui n'allait pas : le fichier de configuration ne portait pas le bon nom.
Je ne comprends pas comment c'est possible, j'ai du faire une fausse manipulation à un moment donné.

Donc il fallait bien mettre le chemin pour le Path dessus et ensuite avoir le fichier de configuration ok.

Je suis désolée du temps perdu.
Merci de ton aide.

closed 2017-01-20 by ColinPitrat

FredCoadou, 2017-01-20:
Colin, je ne sais pas si tu peux voir un sujet fermé....
En fait pour le nom du fuchier de configuration, j'avais pris celui qui est dans le README

```
# Configuration file
 * Where to place your configuration file
     You can modify the default behavior of kalenji_reader by putting a .kalenji_readerrc file in your home directory.
     For example for a user called toto, the configuration file will be:
     > /home/toto/**.kalenjirc**
```

ColinPitrat, 2017-01-20:
Effectivement ! C'est corrigé.