ColinPitrat/kalenji-gps-watch-reader #60: Keymaze 700 trial no detected

bouzin, 2017-02-11:
Hi,
I would like to transfer files from my keymaze 700 trial so I downloaded your pack and tried to install it.
When I use this command : ./kalenji-reader there is an error message : error can't access usb device: -3LIBUSB_ERROR_ACCESS
It seems that my gps watch is not detected.
Do you have a solution please ?
THanks

ColinPitrat, 2017-02-11:
Hi,

This is probably a permission error. One option to verify that is to run kalenji_reader as root.
Another option is to change the permissions on the device (cf https://github.com/ColinPitrat/kalenji-gps-watch-reader/wiki/Frequently-Asked-Questions#libusb-couldnt-open-usb-device-devbususbxxxyyy-permission-denied)

If this doesn't help, you can try the following which may provide more info:
`make debug`
`./kalenji_reader -v`

bouzin, 2017-02-13:
HiThanks a lot. Il was a problem of permission effectively.I run the command as root and it works.Thank you for your job.Best regardsThomas

 Envoyé le : Dimanche 12 février 2017 0h26

ColinPitrat, 2017-02-13:
Hi,
running as root works but the proper fix would be to modify your configuration to change the permissions on the devices. In most distributions, this can be done by copying the file etc/10-kalenji-udev.rules (in git repository) to /etc/udev/rules.d/ (or something similar).

The file is available here: https://github.com/ColinPitrat/kalenji-gps-watch-reader/blob/master/etc/10-kalenji-udev.rules

I'll update the doc stating this.

Regards,
Colin

closed 2017-02-13 by ColinPitrat