ColinPitrat/kalenji-gps-watch-reader #62: OnMove500 pas supportée

TontonManu, 2017-06-07:
Bonjour, je viens d'acquérir la montre Onmove500, et apparemment ce modèle n'est pas encore supporté.
J'ai installé l'application sans erreur apparente (log joint) mais quand je la lance:
```
./kalenji_reader -D OnMove500 -p /media/emmanuel/ONMOVE-500
```
j'ai le retour:
```
Error trying to register device OnMove500: Unknown device
error std::exception
```
(ce qui est normal si OnMove500 n'est pas encore supportée)

J'ai essayé avec "-D OnMove710" et "-D OnMove510" mais les retours sont les mêmes:
```
src/main.cc:292: Configuration parsed
src/main.cc:296: Create output directory '/tmp/kalenji_import'
src/main.cc:373: Registering device
src/main.cc:380: Attaching source to device
src/main.cc:383: Initializing device
src/main.cc:385: Device initialized
src/main.cc:387: Get sessions list
src/main.cc:393: Filter out sessions
src/main.cc:396: Get sessions details
src/main.cc:399: Release device
```
mais aucun fichier ne se trouve dans les dossiers /tmp/kalenji_import/ et /tmp/kalenji_import/logs

J'ai fait une petite marche aujourd'hui pour enregistrer une activité, deux fichiers se trouvent maintenant dans le dossier DATA. Je les joins également.

Merci d'avance !
[onmove500.zip](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/1059170/onmove500.zip)

ColinPitrat, 2017-06-08:
Bonjour,

utiliser -D OnMove200 semble bien fonctionner.
Est-ce que cela fonctionne sans fournir de modèle (sans l'option -D) ?
Sinon, quels sont les vendorId / productId (obtenus avec `lsusb`) ?

Colin

TontonManu, 2017-06-08:
Bonjour, lsusb me donne:
```
Bus 002 Device 025: ID 283e:0024
```
(rien d'autre sur la ligne)

Quand j'utilise -D OnMove200 j'ai en retour:
```src/main.cc:292: Configuration parsed
src/main.cc:296: Create output directory '/tmp/kalenji_import'
src/main.cc:373: Registering device
src/main.cc:380: Attaching source to device
src/main.cc:383: Initializing device
src/main.cc:385: Device initialized
src/main.cc:387: Get sessions list
src/main.cc:393: Filter out sessions
src/main.cc:396: Get sessions details
src/main.cc:399: Release device
```

mais aucun fichier n'est produit dans le répertoire /tmp/kalenji_import/

Quand je ne mets pas l'option 'D':
```
./kalenji_reader -v -p /media/(user)/ONMOVE-500
```
le message est:
```
src/main.cc:292: Configuration parsed
src/main.cc:296: Create output directory '/tmp/kalenji_import'
Can't autodetermine device with a source different from USB. Specify your device using -D option or device= in your configuration file.
```
Bizarre...

ColinPitrat, 2017-06-08:
Le vendorId n'est pas le même que l'OnMove200, c'est pour ça donc l'auto-detection ne marche pas.
Par contre en spécifiant `-D OnMove200` ça devrait marcher.

En compilant en mode debug, tu auras plus d'informations:

`
make clean
make debug
kalenji_reader -v -D OnMove500 -p /media/<user>/ONMOVE-500
`

Pour ma part j'ai le résultat suivant:

`
src/main.cc:292: Configuration parsed
src/main.cc:296: Create output directory '/tmp/kalenji_import'
src/main.cc:373: Registering device
src/main.cc:380: Attaching source to device
src/main.cc:383: Initializing device
src/main.cc:385: Device initialized
src/main.cc:387: Get sessions list
OnMove200: Get sessions list !
Decode summary of session ACT_0000
src/main.cc:393: Filter out sessions
src/main.cc:396: Get sessions details
Retrieve session ACT_0000
src/main.cc:399: Release device
OnMove200: Release (nothing to do)
`

Si les fichiers `ACT_0000.OMD` et `ACT_0000.OMH` sont bien visibles dans `/media/<user>/ONMOVE-500` ça devrait marcher.

TontonManu, 2017-06-08:
Ça fonctionne partiellement. Je peux maintenant générer un fichier .gpx, mais il ne comporte apparemment pas de données d'altitude. La notice indique que ce modèle enregistre bien une altitude, mais que celle-ci est ensuite recalculée sur le site de Decathlon, pour avoir une meilleure précision.

Pour générer le fichier .gpx, j'ai du indiquer le chemin complet vers les fichiers .OMD et .OMH:
```
kalenji_reader -v -D OnMove200 -p /media/<user>/ONMOVE-500/DATA
```
et le résultat donne:
```
src/main.cc:292: Configuration parsed
src/main.cc:296: Create output directory '/tmp/kalenji_import'
src/main.cc:373: Registering device
src/main.cc:380: Attaching source to device
src/main.cc:383: Initializing device
src/main.cc:385: Device initialized
src/main.cc:387: Get sessions list
src/main.cc:393: Filter out sessions
src/main.cc:396: Get sessions details
Retrieve session ACT_0000
Retrieve session ACT_0001
src/main.cc:399: Release device
  Applying filter UnreliablePoints
  Applying filter EmptyLaps
Creating /tmp/kalenji_import/20170607_182500.gpx
Creating /tmp/kalenji_import/20170607_182500.html
Using GoogleMap output requires a Google API Key. You can get one from https://developers.google.com/maps/documentation/javascript/get-api-key
```
Je trouve ensuite le fichier .gpx (à noter qu'il y a aussi un fichier .html, mais qu'il est vide) que je peux ensuite importer dans TurtleSport, par exemple.
Mais dans l'appli, les données d'altitude sont complètement incohérentes (-2147483648m, gloups je suis tombé bien bas). Est-ce que c'est lié au message sur GoogleMap?

ColinPitrat, 2017-06-08:
Je ne me souviens plus si l'OnMove200 ne fournit pas l'altitude ou si c'est juste le code qui ne le supporte pas (je vais jeter un coup d'oeil au code pour vérifier). Il est possible de corriger l'altitude en utilisant GoogleMap en ajoutant un filtre (option `-f`): 

    kalenji_reader -v -D OnMove200 -p /media/<user>/ONMOVE-500/DATA -f FixElevation

D'autre filtres utiles existent pour renseigner des données non fournies par la montre, comme `ComputeInstantSpeed` (qui calcule la vitesse instantanée) ou `ComputeSessionStats` (qui calcule des information pour la session complète: dénivelé positif, négatif, vitesse max, moyenne, durée et distance)

Le message d'erreur à propos de GoogleMap est la raison du fichier HTML vide. Il était possible autrefois d'utiliser des cartes GoogleMap sans API key mais ce n'est malheureusement plus le cas. Pour créer sa propre API key, il suffit de:
 - Avoir un compte google
 - Aller sur: https://developers.google.com/maps/documentation/javascript/get-api-key
 - Cliquer sur "Get a key"
 - Choisir "Create a new project" et le nommer (par exemple "kalenji-reader")
 - Lorsque le résultat s'affiche, copier/coller la chaine de caractères (40 caractères alpha-numériques, quelque chose du genre "VBjwgLpgMXmMggdKJZaH4Yn5cRP4TTLYU9ZuP9cL") dans le fichier de configuration à la valeur "google_api_key"

    google_api_key=VBjwgLpgMXmMggdKJZaH4Yn5cRP4TTLYU9ZuP9cL

TontonManu, 2017-06-08:
Ça a l'air pas mal maintenant.
Sans créer une API key, uniquement avec le filtre FixElevation j'ai eu une erreur "Error: received an error from Google API: "OVER_QUERY_LIMIT"."
Après avoir créé une API key et l'avoir ajoutée dans le fichier de conf (merci pour ton aide!) plus de message d'erreur et les données d'altitude sont bonnes. Il faudra juste que je compare avec le fichier gpx extrait du site de Decathlon, mais ça ne devrait pas varier beaucoup.
Un grand merci!

ColinPitrat, 2017-06-08:
De rien !
Il y a en effet une limite sur le volume de requêtes qui peuvent être faites à l'API Google et j'ai eu l'impression récemment qu'elle avait baissé (dans le passé je pouvais importer une dizaine de sessions sans avoir le problème). L'utilisation d'une API key permet de lever cette limite.

closed 2017-06-08 by ColinPitrat