ColinPitrat/kalenji-gps-watch-reader #64: ONMOVE 500 récemment achetée !

Cobeweb, 2017-06-11:
Bonjour !

Je constate que l'on parle français ici, alors j'en profite!
Je viens d'acheter une montre GPS cardio etc ONMOVE500 (géonaute). bien évidemment, je suis sous Linux (Mageia 5)...
J'ai installé ce super kalenji-gps-watch-reader, mais il me semble que le modèle dont je dispose est trop récent... j'ai invariablement le message suivant:

Error trying to register device OnMove500: Unknown device
error std::exception

Est-ce que vous seriez prêt pour rendre ce modèle compatible? Si c'est le cas, et dans la mesure de mes compétences, je suis prêt à participer !

Bonne fin de journée

ColinPitrat, 2017-06-12:
cf #62 
En spécifiant -D OnMove200 ça devrait marcher.

Je garde le ticket ouvert pour fournir une solution "propre" (gérer plusieurs VendorId/ProductId pour un même type de device).

Cobeweb, 2017-06-12:
Super ! Génial ! Merci beaucoup !
Comme Tontonmanu, je trouve un fichier GPX avec les informations (parcours, distance, temps, et moyenne...) que je souhaitais récupérer (avec un fichier html vide) !
Je me suis aussi fendu d'un petit message à Géonaute pour leur rappeler l'intérêt des formats ouverts et de Linux...

Merci encore pour ce travail !

Docbroke, 2017-10-01:
I tried same for my onmove500 device and I am able to get GPX file, but when I upload that to decathlon coach it fails to recognize heart-rate data.

Sorry if it is already addressed somewhere as I couldn't read French.

ColinPitrat, 2017-10-03:
Previous users of OnMove500 (Cobeweb in this issue and TontonManu in #62) didn't report any issue with heartrate (it doesn't mean there isn't though ;-)).

In the GPX format, there is no "native" support for heartrate so you have to choose a GPX extension to provide it. By default 'gpxdata' extension is used. Maybe Decathlon Coach doesn't recognize it. You can try using 'gpxtpx' instead.
It can be done setting 'gpx_extensions' to either 'gpxtpx' in your [configuration file](https://github.com/ColinPitrat/kalenji-gps-watch-reader/wiki/Configuration-File)

Do you have GPX with heart rate data that you can import in Decathlon Coach ? If this is the case, you can try comparing the content of those files with the ones generated by kalenji_reader (or attach them so that I can check).

Tell me if it helps, otherwise I can have a deeper look.

Docbroke, 2017-10-03:
I am attaching a file created by Decathlon Coach, which seems to use gpxtpx format. As soon as I reach home I will try to convert data using 'gpx_extensions=gpxtpx' and report back.
https://transfer.sh/8BsvJ/20171001_070600.gpx

ColinPitrat, 2017-10-03:
The linked file is created by kalenji_reader and uses gpxdata :-) If the ones exported by Decathlon Coach exports gpxtpx, I expect that it should work to use it to import.
Maybe setting both extensions as the default would make sense. The files would be bigger but this way they should work in all cases ...

Docbroke, 2017-10-03:
uploaded wrong file. Here is the one created by Decathlon Coach

https://transfer.sh/dyaSg/9e88a7e4a8737a72f097.gpx

Also I can confirm now that, gpxtpx works, with Decathlon coach.

desplom, 2018-09-06:
Dear all, first thanks for creating this tool. Then...

I borrowed a Onmove500 from Decathlon (yes, they borrow some material for testing!) so I have few days to test if it works. I compiled the code properly but looks like my computer dos not detect the usb device. I am running Ubuntu 16.04. Running "./kalenji_reader -v -D OnMove200" gives me:

src/main.cc:289: Configuration parsed
src/main.cc:293: Create output directory '/home/usuari/Running'
src/main.cc:309: Source is USB
src/main.cc:313: With transaction logger
src/main.cc:371: Registering device
src/main.cc:378: Attaching source to device
src/main.cc:381: Initializing device
Error: path '' does not exist (check option -p <path> on command line or line path=<path> in configuration file).

Am I supposed to see the device as a normal usb stick so as to be able to tell the path of where the logs are? By now I tried everything: connecting while the device is on or off, checking different usb ports of my laptop but it does not show off...

Any idea? Thanks in advance!

ColinPitrat, 2018-09-07:
The link tells me 'Not Found'.
But if I understand correctly, the problem is fixed ?

closed 2018-09-07 by ColinPitrat

tcharlss, 2019-04-11:
Hi,

In the [Device type section of the wiki](https://github.com/ColinPitrat/kalenji-gps-watch-reader/wiki/Configuration-File#device-type), it says to use « Kalenji » for On Move 500 devices.
However it doesn't work for me, I end up with this error :
```
Erreur de segmentation (core dumped)
```
With « OnMove200 », it works just fine.

ColinPitrat, 2019-09-02:
@tcharlss Thanks for the information. I'll update it. The problem is that they keep reusing the same name for different devices. The original OnMove500 needed Kalenji, I think the model you have is probably a OnMove 500 HRM. The best is to use autodetect for the device type. Does it work well without providing the device type?

tcharlss, 2019-09-03:
Indeed, I have the latest model I think: https://www.decathlon.fr/montre-gps-cardio-onmove-500-id_8391961.html
I removed the `device=OnMove200` line in the config file (this should enable autodetection, right?), but it doesn't work in my case:

> Can't autodetermine device with a source different from USB. Specify your device using -D option or device= in your configuration file.

I'm using the path option to detect the watch:
```
# Mount point
source=Path
path=/media/charles/ONMOVE-500/DATA
```

Anyway, thanks for the explanation. What a strange practice to reuse the same name for different devices!

ColinPitrat, 2019-09-03:
Oh yes, I forgot about it. Indeed for USB mass storage devices, you have to specify the device.