ColinPitrat/kalenji-gps-watch-reader #65: API for Zonefive SportTracks

Eindhoven562, 2017-08-14:
Hello Colin. 

I'am using the kalenji-gps-watch-reader for a long time and I really like it. I have a request (don't know if this is the right place for it). I export the data to a gpx-file which I can import in Zonefive Sporttracks. A direct import in SportTracks (API) would be very usefull. SportTracks is well documented. Would it be possible for you to create an API in the kalenji-gps-watch-reader for SportTracks?

With regards,
Nicolas

ColinPitrat, 2017-08-14:
Hi,

This is definitely the right place for requests :-)

I guess you're talking about https://www.sporttracks.mobi
API is documented here: https://www.sporttracks.mobi/api/doc

It indeed makes sense. It should be easy to add an output (in src/output) that uploads new sessions.
If you know a bit of C++ and want to give it a try, you can send me a pull request. Otherwise I'm not sure when I'll have time to work on it.

We could even add a source (src/source) to import sessions from SportTracks and export them as other formats.

Regards,
Colin

Eindhoven562, 2017-08-14:
Hello Colin,

Yes, that's the programme I'm talking about. I use the pc-version though. I have studied the possibility of writing an API myself, but I don't have the skills to do so. It would be great if you can find some time to work on it. Perhaps I can assist you with some testing. But if you can't it's ok too. Then I will continue the 2-step-import via GPX. 

With regards,
Nicolas

ColinPitrat, 2017-11-07:
Some news about it. Generating the JSON and sending the query is easy to do but the authentication with OAuth2 is a bit more tricky. It will require starting an HTTP server to redirect answers there
I'll check what I can do to do it without drawing too many new dependencies.

Eindhoven562, 2019-10-08:
This issue can be closed. Zonefive Sporttracks announced the EOL (End-Of-Life) for the programme. So an API is not usefull anymore.

closed 2019-10-08 by Eindhoven562