ColinPitrat/kalenji-gps-watch-reader #66: OnMove120

Crashtime, 2017-09-10:
Hey do you think this watch could use this Software? I can help with what is needed.
Cheers.

ColinPitrat, 2017-09-11:
What happens if you try running it ?
If it doesn't work, can you do `lsusb` with your watch plugged to see the vendor ID / product ID ?
Is the device recognized as a USB mass storage (i.e like a USB key) ?

Ventripotent, 2017-11-01:
Hello,

I also have a OnMove120 which is recognized by the computer. I can go in the order and see that the folder "DATA" is not empty.
However when I try to run "kalenji_reader" I get the following message : 
`No known USB device found.`

I tried to do what you said on the 11. of September and i get this : 
`olivier@olivier-UX32A:~/kalenji-gps-watch-reader$ lsusb
Bus 002 Device 004: ID 8087:07da Intel Corp. 
Bus 002 Device 003: ID 04f2:b330 Chicony Electronics Co., Ltd Asus 720p CMOS webcam
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0bda:0139 Realtek Semiconductor Corp. RTS5139 Card Reader Controller
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 283e:001f  
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

Do you have any idea what I could do to make it work. I'm really willing to help, but I started with Linux yesterday and I'm still struggling with the terminal ... 

Thank you in advance,
Best Regards

ColinPitrat, 2017-11-01:
OK, so vendorId = 283E and productId = 001F.

If I understand correctly, it's recognized as a USB mass storage ? What kind of files do you find in it ?

You can try running `./kalenji_reader -D OnMove100` to see if it works. Otherwise, you can also try OnMove200 and OnMove500. Double check the output makes sense if any of these work.

If none works, you can attach the files to this issue and I'll have a look at them.

Ventripotent, 2017-11-01:
Once again thanks for answering so quick !
None of the three commands seem to work ... Here is what I get back : 
`olivier@olivier-UX32A:~/kalenji-gps-watch-reader$ ./kalenji_reader -D OnMove100
error USB device not found !
olivier@olivier-UX32A:~/kalenji-gps-watch-reader$ ./kalenji_reader -D OnMove200
Error: path '' does not exist (check option -p <path> on command line or line path=<path> in configuration file).
error std::exception
olivier@olivier-UX32A:~/kalenji-gps-watch-reader$ ./kalenji_reader -D OnMove500
Error trying to register device OnMove500: Unknown device
error std::exception
`
As you proposed, i attach the files I found in the order "DATA" of the order OnMove120. I also add two screenshots to be as clear as possible.

Thanks for your help
Best regards

[DATA.tar.gz](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/1435000/DATA.tar.gz)

[image]
[image]

ColinPitrat, 2017-11-02:
Oh yes, my bad. You have to provide the path to the device in the command (/media/ONMOVE-120/DATA in my example, not sure if it's exactly that).
The OMD / OMH files are the names used by the OnMove200 so you can try this:
`./kalenji_reader -D OnMove200 -p /media/ONMOVE-120/DATA`

Ventripotent, 2017-11-07:
Hi ! Sorry I didn't make it this week with the training so i didn't try the command before today ...
Here is what I get : 
`olivier@olivier-UX32A:~/Documents/kalenji-gps-watch-reader$ ./kalenji_reader -D OnMove200 -p /media/olivier/ONMOVE-120/DATA
Retrieve session ACT_0001
  Applying filter UnreliablePoints
  Applying filter EmptyLaps
Creating /tmp/kalenji_import/20171107_121500.gpx
Creating /tmp/kalenji_import/20171107_121500.html`

:D :+1: 

So I think you can proudly once more say that your code is also working for the watch OnMove 120 !
I'm unable to close the topic because I didn't create it but problem solved 

Thank you very much !

ColinPitrat, 2017-11-07:
Cool. I'm working on making it work out of the box (which means having the same device driver auto-detect multiple devices instead of only one).

ColinPitrat, 2019-05-17:
Closing this issue, will track auto detection of devices sharing the same vendor ID / device ID separately.

closed 2019-05-17 by ColinPitrat