ColinPitrat/kalenji-gps-watch-reader #67: Impossible to run the command "make"

Ventripotent, 2017-10-31:
Hi ! 

First of all thank you for your work, I saw a lot of posts on forums referring to your work as the ideal solution for linux users !
Unfortunately i'm unable to run the command run as supposed to ... i get the following error but i don't understand it (i'm a real beginner with linux but i need my running watch to work with it ^^'). Here is the error i get : 

olivier@olivier-UX32A:~/Téléchargements/kalenji-gps-watch-reader$ make
find: «test/unit»: Aucun fichier ou dossier de ce type
find: «test/unit»: Aucun fichier ou dossier de ce type
Error: missing dependency libusb-1.0. Try installing libusb development package (e.g: libusb libusb-1 libusb-1.0.0-dev ...)
Makefile:72 : la recette pour la cible « check_deps » a échouée
make: *** [check_deps] Erreur 1

Does anyone understand it and could help me ? 

Thank you in advance :)
Ventripotent

ColinPitrat, 2017-10-31:
The important part of the message is: `Error: missing dependency
libusb-1.0. Try installing libusb development package (e.g: libusb libusb-1
libusb-1.0.0-dev ...)`

It says that you need to install libusb-dev package. This should be
available in your distribution's package manager. For example for debian or
ubuntu, `apt-get install libusb-1.0.0-dev`

Package provide ways to search for packages in case none of the suggested
names is the right one. For example: `apt-cache search libusb`. Note that
you'll need the development package (providing headers, usually has the
-dev suffix) not just the library one.

Once done, you may encounter a similar message for libxml and libcurl.

Regards,
Colin

<https://github.com/ColinPitrat/kalenji-gps-watch-reader/issues/67>, or mute
the thread

Ventripotent, 2017-10-31:
Ok ! It was actually pretty easy ... I'm sorry I had to bother you for such a little problem !
Indeed I had to download libxml and libcurl as well.
It was a perfect concrete example for me to learn to use the command sudo ! 
First day on linux and already so much learned :) It's a real pleasure to find such a community ready to help !

I'll have to go running tomorrow to check if I'm able to export the file :)
Thanks again for your help !

Best regards

closed 2017-10-31 by Ventripotent