ColinPitrat/kalenji-gps-watch-reader #71: Import keymaze trial 700

bouzin, 2018-04-15:
Bonjour je n'arrive plus à importer mes séances depuis mon keymaze trial 700.
Voici le message que j'ai : 
 
`tom@tom-Vostro-1700:~/Documents/Course` à pied/kalenji-gps-watch-reader$ sudo ./kalenji_reader
[sudo] Mot de passe de tom : 
Using device type Keymaze 700 Trial
error on USB operation code -7 - 11 LIBUSB_ERROR_TIMEOUT``

Merci pour votre aide
Tom

ColinPitrat, 2018-04-15:
Can you try running with option `-v` with a version built in debug from a fresh checkout ?

bouzin, 2018-04-16:
I'm very beginner with linux so it is difficult for me. I've tried you suggested me but still have the same problem.
With -v I've this message : 

> GNU Make 4.1
Construit pour i686-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
Licence GPLv3+ : GNU GPL version 3 ou ultérieure <http://gnu.org/licenses/gpl.html>
Ceci est un logiciel libre : vous êtes autorisé à le modifier et à la redistribuer.
Il ne comporte AUCUNE GARANTIE, dans la mesure de ce que permet la loi.

What should I do ?

Thank you

ColinPitrat, 2018-04-16:
The -v is fo kalenji_reader:
```
make debug
./kalenji_reader -v
```

Did it ever work or is it the first time you try ?

bouzin, 2018-04-16:
Thank you
I obtained that : 
`tom@tom-Vostro-1700:~/Documents/Course à pied/kalenji-gps-watch-reader$ make debug
find: «test/unit»: Aucun fichier ou dossier de ce type
find: «test/unit»: Aucun fichier ou dossier de ce type
[ -d googletest ] || git clone https://github.com/google/googletest.git
/bin/sh: 1: git: not found
Makefile:43 : la recette pour la cible « gtest » a échouée
make: *** [gtest] Erreur 127
tom@tom-Vostro-1700:~/Documents/Course à pied/kalenji-gps-watch-reader$ ./kalenji_reader -v
src/main.cc:292: Configuration parsed
src/main.cc:296: Create output directory '/tmp/kalenji_import'
src/main.cc:312: Source is USB
src/main.cc:316: With transaction logger
src/main.cc:336: Auto-detecting device
src/main.cc:364: Auto-detected Kalenji
src/main.cc:373: Registering device
src/main.cc:380: Attaching source to device
src/main.cc:383: Initializing device
error can't access usb device: -3LIBUSB_ERROR_ACCESS
`

bouzin, 2018-04-16:
Sorry it is not very easy to read.

tom@tom-Vostro-1700:~/Documents/Course à pied/kalenji-gps-watch-reader$ make debug
find: «test/unit»: Aucun fichier ou dossier de ce type
find: «test/unit»: Aucun fichier ou dossier de ce type
[ -d googletest ] || git clone https://github.com/google/googletest.git
/bin/sh: 1: git: not found
Makefile:43 : la recette pour la cible « gtest » a échouée
make: *** [gtest] Erreur 127
tom@tom-Vostro-1700:~/Documents/Course à pied/kalenji-gps-watch-reader$ ./kalenji_reader -v
src/main.cc:292: Configuration parsed
src/main.cc:296: Create output directory '/tmp/kalenji_import'
src/main.cc:312: Source is USB
src/main.cc:316: With transaction logger
src/main.cc:336: Auto-detecting device
src/main.cc:364: Auto-detected Kalenji
src/main.cc:373: Registering device
src/main.cc:380: Attaching source to device
src/main.cc:383: Initializing device
error can't access usb device: -3LIBUSB_ERROR_ACCESS

ColinPitrat, 2018-04-16:
OK, the `make debug` fails because it doesn't find git (isn't it installed ? How do you clone the repository then ?).
There will be less logs because of that but it may not be a big issue, we can have a first look without it. However, your error here is an access error, not the same as before, because you didn't run with sudo.

Can you try:
`sudo ./kalenji_reader -v`

bouzin, 2018-04-16:
it done that : 
src/main.cc:292: Configuration parsed
src/main.cc:296: Create output directory '/tmp/kalenji_import'
src/main.cc:312: Source is USB
src/main.cc:316: With transaction logger
src/main.cc:336: Auto-detecting device
src/main.cc:364: Auto-detected Kalenji
src/main.cc:373: Registering device
src/main.cc:380: Attaching source to device
src/main.cc:383: Initializing device
Using device type Keymaze 700 Trial
src/main.cc:385: Device initialized
src/main.cc:387: Get sessions list
src/main.cc:393: Filter out sessions
src/main.cc:396: Get sessions details
error on USB operation code -7 - 11 LIBUSB_ERROR_TIMEOUT

bouzin, 2018-04-17:
I don't remember if i installed git. I think not.
After I transfered my running sessions from device, I imported its to pytrainer from the directory '/tmp/kalenji_import'.
I use kalenji reader since more one year without problem. This week-end it didn't work.

ColinPitrat, 2018-04-17:
Did you update it ? Otherwise it means something happened with the watch, or some unfinished business between the watch and the software. The strange part is that it manages to do the list sessions part (so it seems it communicates with the watch) but fails on the retrieve sessions.

It doesn't seem to display a list of sessions after the get sessions list. Is there any on the watch ? Can you attach the logs (should be in /tmp/kalenji_import/logs) ?

bouzin, 2018-04-17:
I did some update on my computer last days but I c'an't tell you if git has been updated.
Yes there are sessions on the watch.
What do you mean by attach the logs ?

ColinPitrat, 2018-04-17:
You should have a file under /tmp/kalenji_import/logs that contains the exchange between kalenji_reader and the watch. Can you attach it to this issue ?

bouzin, 2018-04-17:
Ok I'll try when I will be back home (tonight - french hour).

bouzin, 2018-04-17:
[kalenji_reader_20180416_225412.log](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/1920632/kalenji_reader_20180416_225412.log)

bouzin, 2018-05-14:
Hello Colin,
I still not success to transfer my sessions. I have installed git and mahe a new installation of kalenji-reader but it doesn't work.
There is this error message : 
error can't access usb device: -3LIBUSB_ERROR_ACCESS
May you help me please ?
Thomas

ColinPitrat, 2018-05-14:
From your log, it seems that the watch stop answering in the middle of the first session, after receiving date for ~30 minutes (over a ~45 minutes session).
You can try 3 things:
 - retry importing multiple times and attach the logs corresponding to the attempts, to see if it always stop at the same point
 - import with the official software to verify it works
 - setting 'import=ask' in your configuration file (on linux ~/.kalenji_readerrc - see https://github.com/ColinPitrat/kalenji-gps-watch-reader/wiki/Configuration-File#default-import-behavior) and then try to import other sessions, to see if it's an issue with this particular session

bouzin, 2018-05-15:
It doesn't work. I tried a new installation and at th end there is this message : 
/usr/bin/ld : ne peut ouvrir le fichier de sortie test/unit/unit_tester : Aucun fichier ou dossier de ce type
collect2: error: ld returned 1 exit status
Makefile:86 : la recette pour la cible « unit_test » a échouée
make: *** [unit_test] Erreur 1

bouzin, 2018-05-15:
It works !!!!!
The problem was the first session on the watch. I deleted it and then I tried again and it works.
Thank you for all !!!

closed 2020-06-15 by ColinPitrat