ColinPitrat/kalenji-gps-watch-reader #73: Ultrasport NavRun 600 / OnMove100 incorrect tcx and gpx data output

knedlyk, 2019-03-07:
Hi, I tried to get data from Ultrasport NavRun 600, which is basically rebranded version of OnMove100. First of all I should mention, that Onconnect recognize watch as OnMove100 and correctly save tracks from device with no issues. But I use archlinux everywhere and don't want to switch between archlinux and windows on my notebook. I'm using last git version, build on archlinux. By default kalenji_reader recognize watch as PylePGSPW1, but produces error, so I changed device to OnMove100 in configuration file.  So, facts are:
1. Output from lsusb: 
`ID 10c4:ea61 Cygnal Integrated Products, Inc. CP210x UART Bridge`
2. TCX output produces a file with the only one trackpoint:
```
<?xml version="1.0" encoding="utf-8"?>
<TrainingCenterDatabase xmlns="http://www.garmin.com/xmlschemas/TrainingCenterDatabase/v2" xmlns:xsi="http://www.w3.org/2001/XM
LSchema-instance" xsi:schemaLocation="http://www.garmin.com/xmlschemas/ActivityExtension/v2 http://www.garmin.com/xmlschemas/Ac
tivityExtensionv2.xsd http://www.garmin.com/xmlschemas/TrainingCenterDatabase/v2 http://www.garmin.com/xmlschemas/TrainingCente
rDatabasev2.xsd">
 <Activities>
  <Activity Sport="Running">
   <Id>2019-03-07T11:02:38Z</Id>
  </Activity>
 </Activities>
</TrainingCenterDatabase>
```
3. GPX ouput produces files with messed time stamps, example is below:
```
<?xml version="1.0"?>
<gpx version="1.1"
     creator="Kalenji Reader"
     xmlns="http://www.topografix.com/GPX/1/1"
     xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
     xmlns:gpxtpx="http://www.garmin.com/xmlschemas/TrackPointExtension/v1"
     xsi:schemaLocation="http://www.topografix.com/GPX/1/1
                          http://www.topografix.com/GPX/1/1/gpx.xsd">
  <metadata>
    <name>No name</name>
    <time>2019-03-07T11:02:38Z</time>
  </metadata>
  <trk>
    <trkseg>
      <trkpt lat="50.083568" lon="19.853195" >

        <time>2019-03-18T06:21:54.1430Z</time>
        <extensions>
        </extensions>
      </trkpt>
      <trkpt lat="50.083722" lon="19.853115" >

        <time>2019-03-09T23:48:38.000Z</time>
        <extensions>
        </extensions>
      </trkpt>
      <trkpt lat="50.08378" lon="19.853082" >

        <time>2019-03-17T13:42:38.000Z</time>
        <extensions>
        </extensions>
      </trkpt>
      <trkpt lat="50.083797" lon="19.853015" >

        <time>2019-03-12T02:42:38.000Z</time>
        <extensions>
        </extensions>
      </trkpt>
      <trkpt lat="50.083748" lon="19.852943" >

        <time>2019-03-08T01:40:38.000Z</time>
        <extensions>
        </extensions>
      </trkpt>
      <trkpt lat="50.083692" lon="19.852912" >

        <time>2019-03-16T00:38:38.000Z</time>
        <extensions>
        </extensions>
      </trkpt>
      <trkpt lat="50.08365" lon="19.852878" >

        <time>2019-03-08T15:38:38.000Z</time>
        <extensions>
        </extensions>
      </trkpt>
      <trkpt lat="50.083585" lon="19.85289" >

        <time>2019-03-10T12:37:38.000Z</time>
        <extensions>
        </extensions>
      </trkpt>
      <trkpt lat="50.083522" lon="19.852902" >

        <time>2019-03-15T02:36:38.000Z</time>
        <extensions>
        </extensions>
      </trkpt>
      <trkpt lat="50.083445" lon="19.852865" >

        <time>2019-03-09T08:36:38.000Z</time>
        <extensions>
        </extensions>
      </trkpt>
      <trkpt lat="50.08337" lon="19.852823" >

        <time>2019-03-15T13:35:38.000Z</time>
        <extensions>
        </extensions>
      </trkpt>
      <trkpt lat="50.08324" lon="19.852753" >

        <time>2019-03-11T03:35:38.000Z</time>
        <extensions>
        </extensions>
      </trkpt>
      <trkpt lat="50.083192" lon="19.852725" >

        <time>2019-03-10T00:35:38.000Z</time>
        <extensions>
        </extensions>
      </trkpt>
```
Please take a look on time. Instead of 2019-03-07 it gives a different time like 2019-03-10, 2019-03-15, etc. This is not a case of onconnect, it gives correct time for all trackpoints. 

I suggest it is problem of firmware or code. Please tell me what info should I provide to help solve this problem. Log file is attached.
[kalenji_reader_20190307_133604.log](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/2941230/kalenji_reader_20190307_133604.log)
Regards,
Yarema

ColinPitrat, 2019-05-17:
I had a look at the log file attached.
At a high level, it does look similar to the OnMove100 format: similar initialization sequence, same overall format with line ending with ff for header, fd for session header and fa for data.

But the content clearly doesn't match. It may well be the same protocol with a different version, or a completely different protocol. Considering the protocol is not documented and there are many differences everywhere, it's hard to tell.

The simplest is probably to create a separate device NavRun600 forked from OnMove100 and to reverse engineer it. Can you provide some log files (like you did) with the corresponding GPX file? For this you'd need to import once with kalenji_reader and once with Onconnect ...

ColinPitrat, 2019-09-13:
If anybody passing here and who has a Ultrasport NavRun600 is interested in helping support this device, they can provide GPX from OnConnect and import logs from kalenji using OnMove100 driver.
I'm happy to support if help is needed.