ColinPitrat/kalenji-gps-watch-reader #76: OneMove200 ne se monte plus

FredCoadou, 2019-08-03:
Bonjour, ma montre OneMove200 ne se monte plus sur mes ordinateurs.
J'ai un ordinateur qui tourne sous Ubuntu et un autre sous Debian.
Je branche la montre mais elle n'aparait pas dans mon gestionnaire de dossier, comme auparavant et donc lorsque je lance votre script il ne fonctionne pas...
Si vous pouviez m'aider...
Merci

ColinPitrat, 2019-08-03:
C'est de l'USB Mass Storage. Une clef USB fonctionne-t-elle bien ? Si oui, la montre est probablement endommagée. Tester quand même sur un autre ordinateur pour vérifier.

FredCoadou, 2019-08-03:
C'est bien ce que je craignais... Ca fait ça sur deux ordinateurs différents avec OS différents... Je vais essayer de transférer mes données via bluethooth sur mon smartphone... Merci !

closed 2019-08-03 by FredCoadou