ColinPitrat/kalenji-gps-watch-reader #77: Issue when exporting to gpx file

zaurelzo, 2020-05-16:
Hi colin,
Thank you for your awesome work. I own a kalenji watch (on move 220) and since your project does not support yet this watch, I changed my local config in order to use the on move 200 config. I am using the kalenji reader in order to generate a gpx file of my bike ride and import it to strava. So far, it worked pretty well, but I am experiencing an interesting issue. Sometimes, when exporting my ride to gpx, the kalenji reader add a last point with a bad geolocation. I manually discard this last point before uploading my file to strava, but I was wondering if you have any idea of why this last bad point is generated.

My kalenji-reader config
```
source=Path
path=/media/user/ONMOVE-220/DATA
device=OnMove200
filters=FixElevation,ComputeInstantSpeed
gpx_extensions=none
```

ColinPitrat, 2020-05-17:
I can have a look but it would probably be useful if you can provide logs and an example of GPX file with the issue:
https://github.com/ColinPitrat/kalenji-gps-watch-reader/wiki/Configuration-File#log-transactions

You just need to set log_transactions=yes in your configuration to produce the logs of the dump.

BBllrd, 2021-03-21:
I have the same issue with Onmove 500, which have almost the same data structures, with blocs of 3 lines, 2 lines for points coordinate, and 1 line with metadata about the 2 points.
The issue is when at the end of the file, there is only one point, le second line is red like a point but it is metadata about the first line.
I dealed with that when I wrote Onmove 500 support : https://github.com/ColinPitrat/kalenji-gps-watch-reader/issues/83