ColinPitrat/kalenji-gps-watch-reader #78: unable to import from onmove200

b3b00, 2020-06-14:
i am unable to import a single activity from my geonaute on move 200.
my conf : linux mint 19.3 
kenel Linux linux 5.3.0-53-generic #47~18.04.1-Ubuntu SMP Thu May 7 13:10:50 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

verbose output
```
@linux:~/dev/kalenji-gps-watch-reader$ ./kalenji_reader -o GPX -d .  ~/.kalenji_readerrc -p /media/olivier/ONMOVE200/ -D OnMove200 -v
src/main.cc:289: Configuration parsed
src/main.cc:293: Create output directory '.'
src/main.cc:371: Registering device
src/main.cc:378: Attaching source to device
src/main.cc:381: Initializing device
src/main.cc:383: Device initialized
src/main.cc:385: Get sessions list
src/main.cc:391: Filter out sessions
Sessions available for import:
List of sessions to import (space separated - 'all' to import everything - 'new' to import only new sessions): 
all
src/main.cc:394: Get sessions details
src/main.cc:397: Release device
olivier@linux:~/dev/kalenji-gps-watch-reader$ ^C

```

i have 2 files in onmove /data folder : 

-  ACT_0000.OMD

- ACT_0000.OMH

I can provide more information if you want

b3b00, 2020-06-14:
dmesg dump :
````
[  629.855266] ata3: EH complete
[  629.855522] ata3.00: Enabling discard_zeroes_data
[  658.982209] usb 1-1.1: USB disconnect, device number 52
[  916.176394] usb 1-1.1: new full-speed USB device number 54 using ehci-pci
[  916.454028] usb 1-1.1: config 1 interface 0 altsetting 0 endpoint 0x83 has an invalid bInterval 0, changing to 10
[  916.455023] usb 1-1.1: New USB device found, idVendor=15a2, idProduct=0300, bcdDevice= 0.02
[  916.455025] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  916.455026] usb 1-1.1: Product: OnMove-200
[  916.455028] usb 1-1.1: Manufacturer: FREESCALE SEMICONDUCTOR INC.
[  916.455029] usb 1-1.1: SerialNumber: BAD STRING INDEX
[  916.455760] cdc_acm 1-1.1:1.0: ttyACM0: USB ACM device
[  916.799937] usb 1-1.1: USB disconnect, device number 54
[  921.496308] usb 1-1.1: new full-speed USB device number 55 using ehci-pci
[  921.727561] usb 1-1.1: New USB device found, idVendor=283e, idProduct=0009, bcdDevice= 0.01
[  921.727565] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  921.727567] usb 1-1.1: Product: ONMOVE-200     
[  921.727568] usb 1-1.1: Manufacturer: FREESCALE SEMICONDUCTOR INC.
[  921.727570] usb 1-1.1: SerialNumber: 004B0007401734474E45
[  921.732487] usb-storage 1-1.1:1.0: USB Mass Storage device detected
[  921.733592] scsi host7: usb-storage 1-1.1:1.0
[  922.761214] scsi 7:0:0:0: Direct-Access     ONMOVE   -200             0001 PQ: 0 ANSI: 4
[  922.765235] sd 7:0:0:0: Attached scsi generic sg3 type 0
[  922.765807] sd 7:0:0:0: [sdc] 510 4096-byte logical blocks: (2.09 MB/1.99 MiB)
[  922.766423] sd 7:0:0:0: [sdc] Write Protect is off
[  922.766426] sd 7:0:0:0: [sdc] Mode Sense: 00 00 00 00
[  922.767054] sd 7:0:0:0: [sdc] Asking for cache data failed
[  922.767060] sd 7:0:0:0: [sdc] Assuming drive cache: write through
[  922.793995]  sdc: sdc1
[  922.797244] sd 7:0:0:0: [sdc] Attached SCSI removable disk

```

ColinPitrat, 2020-06-14:
It doesn't find any session to import. I should add a log to say in which directory it's looking, I suspect this may be the issue. IIUC, the files are in a data subdirectory, did you try adding it to the path?

b3b00, 2020-06-14:
ok so you re saying that the -p arg must target the DATA dir where the files are ? i thought it simply is the mount point of the mass storage. i will try later using -p /media/olivier/ONMOVE200/DATA

b3b00, 2020-06-15:
its working fine with the full path to omh and omd files. thank you for this great piece of software

closed 2020-06-15 by b3b00