ColinPitrat/kalenji-gps-watch-reader #79: PGSPW1 problem to connect

goga1912, 2020-07-12:
I have PGSPW1 and try to get data from it.
I installed winusb driver successfully, but when i try to get data by kalenji_reader, i get mistake:
error on USB operation code -7 - 0 LIBUSB_ERROR_TIMEOUT 
=> 54 5a 01 01 00 00 ff ff 
I try it on two PC: windows 10 and windows 7 and hade same result.

How can i fix this problem?

ColinPitrat, 2020-07-14:
There's an issue reported on PGSPW1 from other people:
https://github.com/ColinPitrat/kalenji-gps-watch-reader/issues/16#issuecomment-79454938

It seems the same watch name has been used for different models which means some will work and some will not. Someone reported it in the past and provided an implementation that worked for them in the comment above. However, as I didn't want to break the existing, I never integrated it because I would need a way to differentiate it.
You can try replacing your PylePGSPW1.cc file by the one from the comment and see if it works. If it does, tell me and I'll integrate it with a new device name.

ColinPitrat, 2020-07-14:
I just realized the issue has been imported from googlecode a long time ago and the file was lost in the process ...

goga1912, 2020-07-14:
Any chance of solving the problem?

ColinPitrat, 2020-07-15:
Unfortunately, I don't think I'll be able to recover the file that was attached to the issue.
This would need to be reimplemented based on a dump obtained from your watch under windows.
https://github.com/ColinPitrat/kalenji-gps-watch-reader/wiki/Produce-dumps

goga1912, 2020-07-15:
[dump.zip](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/4926956/dump.zip)

ColinPitrat, 2020-07-16:
Thanks.
There are 2 files: dump_with_watch & dump_without_watch.
What is the one without watch?
For the one with watch, you confirm you uploaded a session while the watch was plugged right?

goga1912, 2020-07-16:
dump_without_watch - no session with watch
dump_with_watch - session with watch

goga1912, 2020-07-25:
What news?

ColinPitrat, 2020-07-25:
So far I only had a quick look but couldn't find any resemblance with the previous Pyle PGSPW1 I wrote support for. I didn't clearly identify the content of the session in it. Could you send me the GPX of the session so that I can compare the data with the USB dump ?

goga1912, 2020-07-27:
Unfortunately, I can't make a successful read with a gpx file: I have problems with the SportTracks program right now, so I want to use your program. Therefore, I can only provide old GPX data. Maybe this will help.
[Template.zip](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/4983595/Template.zip)

goga1912, 2020-08-09:
So?

ColinPitrat, 2020-08-09:
Sorry, I missed your previous answer.
I had a look at the file but they are a bit weird. There are many interrupts but almost no data exchanged.
Now that I see your previous comment, I'm wondering, did you transfer sessions while you captured the USB traffic? Or was the watch just plugged but no transfer done?

goga1912, 2020-08-10:
Attached a dump at the time of exchange via the SportTrack program.
[dump_sporttrack.zip](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/5051986/dump_sporttrack.zip)

ColinPitrat, 2020-08-14:
OK, the protocol is similar to the PylePGSPW1 that is already implemented.
The differences I found are:
 - first byte of queries & response is 0x56 instead of 0x54
 - data request ends with 0xWXYZ where 0xWXYZ is the number of the request in little endian starting at 0 (in current version it's always 0xffff)
 - the response size (bytes 4 & 5 in current version) are the number of 8 bytes lines, not the number of bytes
 - the data size (byte 12 & 13 in current verion) is not here anymore (it was redundant with the response size anyway)
 - there are some lines with only 0xff. It seems they should be ignored
 - sessions still start with 6 * 0xff followed by session number in little endian (starting at 0)
 - laps header still start with 2 * 0xff followed by the timing of the lap
 - some lines start with 1 * 0xff, unsure what they are ...

Could you provide me some details about the sessions?
If the rest of the protocol didn't change too much, it seems there are 4 sessions:
 - 2016-06-08 at 0:00:35
 - 2018-06-30 at 6:54:14
 - 2018-08-05 at 4:57:42 
 - 2019-04-27 at 7:22:05
That are all in Russia, approximately 200km east from Moscou. Is that correct?

And also about the watch? What data do you have about the sessions/laps directly on the watch? Do you have heartrate info? Or only latitude, longitude and elevation?

goga1912, 2020-08-14:
By session: there should be 1 session for the previous year (2019), and about 9 sessions for this year (2020). I don't remember last year, but this year: May, June, July. 
As for latitude / longitude - yes, everything is correct, about 200 km from Moscow. 
Additional information about the session: heart rate is about 130-160, the total session time is about an hour, session time - morning time (about 07:00-09:00 GMT+3), weekday - Saturday or Sunday.

goga1912, 2020-08-14:
Forgot. Not all data was counted through the SportTrack program, so the sessions may not really be all.

ColinPitrat, 2020-08-14:
This last comment is interesting. We may need multiple iterations then to understand the protocol better.
When to stop asking for more data is one of the thing that is not clear to me. We can try a version that will keep asking more data and see what happens.

ColinPitrat, 2020-08-14:
Does the following look like a session you could have made? (it's a round trip done 2 times):
[image]

ColinPitrat, 2020-08-14:
I just committed a PylePGSPW1_v2 device that should manage to get something out of your watch.

After putting in your configuration file the following line:

```
import=all
log_transactions=yes
filters=none
outputs=GPX
```

Can you try something like:

```
./kalenji_reader -v -D PylePGSPW1_v2 
```

This will hopefully manage to import something, create some GPX file and a logs directory with dumps of the transaction in it. Could you please attach the logs to the issue?

Thanks!

goga1912, 2020-08-14:
I don’t understand how can I get the Exe file? I need to compile? How to do this?

ColinPitrat, 2020-08-14:
Ah, I don't have my machine with the setup to do a windows build with me currently. Let me try to setup something quickly on AppVeyor, I thought I had done it already but it was for another project.

In the meantime, does the screenshot in the previous comment looks correct? (is this the place where you've been running?)

goga1912, 2020-08-14:
Yes that's right

ColinPitrat, 2020-08-14:
You can try the latest release here:
https://github.com/ColinPitrat/kalenji-gps-watch-reader/releases/tag/latest

There may be some missing DLL that you'll have to download. I have to work on it to manage to include them in the archive.

goga1912, 2020-08-15:
I tried to get data with a new program, the result is below:
D:\kalenji-gps-watch-reader>kalenji_reader -v -D PylePGSPW1_v2
src/main.cc:289: Configuration parsed
src/main.cc:293: Create output directory 'd:\kalenji-gps-watch-reader\import'
src/main.cc:309: Source is USB
src/main.cc:313: With transaction logger
src/main.cc:371: Registering device
src/main.cc:378: Attaching source to device
src/main.cc:381: Initializing device
error USB device not found !

In this case, the log is missing.
I ran the previous version and the log I received:
 => 54 5a 01 01 00 00 ff ff 
Note that the driver is installed (see srceenshot)
[image]

ColinPitrat, 2020-08-15:
Oh yeah, silly me, I changed the deviceId by mistake.
This is now fixed.
New binary still at the same place:
https://github.com/ColinPitrat/kalenji-gps-watch-reader/releases/tag/latest

goga1912, 2020-08-15:
I get correctly gpx sessions for the last year: it looks like the truth. Only the time is apparently without taking into account GMT, because local time turns out the training was at 6 o'clock in the morning, but in fact it was around 9 o'clock. 
Perhaps this is due to the fact that there was a glitch in the browser with data saving, but can they be restored?
Still tried to count Fitlog, but: 
1. In no there is no data on heart rate. 
2. I tried to import it to SportTrack - not successfully (see screenshot).
[image]
[Import.zip](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/5078319/Import.zip)

goga1912, 2020-08-15:
If there is a glitch in the clock, how can I clear my memory? In the SportTrack program, I can do this after successfully uploading data.

ColinPitrat, 2020-08-15:
I had a bug on my side so it only imported the first session. It was an interesting finding though as it allowed me to verify that the two last bytes of the "data" query actually control the page of data that is returned.

One thing that is not clear to me (and that I want to clarify from the logs I'm asking you, is how to know when all the data has been retrieved. From your previous comment, I suspect SportTrack ended the import to early, that's one of the things I'd like to verify.

The heartrate data is not yet supported because it wasn't available in the previous version of the watch that I supported. However I found where it is and it should be easy to decode.

I also suspect that the elevation (altitude) data is wrong. The previous watch I supported applied an offset and I think it's also the case here but with a different offset value. Do you have the elevation data for a session visible on the watch? or in SportTrack? (the watch would be better because SportTrack may correct it).
Wrong elevation data is not necessarily a big issue as there's a FixElevation filter in kalenji_reader that queries GoogleMap.

For the suppression of sessions, I would need a dump of USB traffic while SportTrack is doing it to see how it's done. I can't guess it without it. WE can try to do it once we've successfully tackle the import.

Can you retry an import with the last update that I made? This should import more sessions. I'm still interested in the same information (log files and GPX/fitlog output).

goga1912, 2020-08-15:
SportsTrack reads only the initial data, and then an error occurs, because of which I can't read all data: apparently there is some incorrect record, because of which I can not fully read all the sessions (see screenshot)
[image]
The elevation (altitude) data is really wrong. I always corrected it in the SportsTrack program.

I read the data again and attached it. More sessions were readed. But not all, because there should be data for June and early July. Although perhaps because of memory overflow, they somehow cleared. Only it is strange that the records were cleared for this year, and not the previous years.
[Import.zip](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/5078764/Import.zip)

ColinPitrat, 2020-08-23:
You mean in the software in which you open the GPX file? Which one do you use?

goga1912, 2020-08-28:
Above, I sent an archive with read files.
If we take for example the file 20200809_050610.fit, then the time is indicated there 2020-08-09T02: 06: 10Z
So, So the time is valid with GMT, but wrong.

ColinPitrat, 2020-08-29:
Ah OK, so the filename is the correct GMT time and the content has an incorrect one.
But actually, the filename should use the local time and the content have the GMT time, so the issue is that I read the time from the watch as if it was localtime when it's actually GMT.

This is now fixed. You should have the session named 20200809_080610.fit (assuming the local time is really 8:06) and the content should have the GMT time 2020-08-09T05:06:10Z.

goga1912, 2020-08-30:
When you try to get data from the watch, you get this error:
[image]

ColinPitrat, 2020-08-30:
Can you translate it? As it's a message box, I suspect it's an issue on the USB driver side.
 Did you change anything recently in your setup? Installed new drivers maybe?

goga1912, 2020-08-30:
Unable to continue code execution because the system did not detect libzstd.dll
The previous version runs without errors

ColinPitrat, 2020-08-30:
The missing DLL has been added and I also added a test to ensure this doesn't happen again.

goga1912, 2020-08-31:
Everything is good. Thanks very much.

goga1912, 2020-09-05:
Everything work.

closed 2020-09-05 by goga1912