ColinPitrat/kalenji-gps-watch-reader #83: Onmove500HR support

BBllrd, 2021-03-19:
Bonjour,

Merci beaucoup pour cet outil qui permet d'utilser nos montres Kalenji. 
Je cours avec une OnMove500HR et j'utilise pour l'instant l'option "-D OnMove200", mais je ne récupère pas l'info de l'altitude, qui est enregistrée par la OnMove500 mais pas la OnMove200.
Je ne suis pas un grand dévelopeur mais je suis reparti du fichier ./device/OnMove200.cc et j'ai créer un fichier OnMove500HR.cc et .h
J'ai changé les noms de tous les objets Onmove200 en OnMove500HR et j'ai étudié mes fichiers binaires en repartant de ce qui avait déjà été fait pour la 200. !!!

J'ai mis en commentaires beaucoup d'infos sur ce que j'ai compris des fichiers binaires. Pour faciliter un travail futur ???

Je vous livre le fichier source et quelques fichiers OMH et OMD pour test. Chez moi ça compile et l'export GPX fonctionne correctement.
J'ai ajouté un dossier de test mais évidement il sort avec succès car le résultat attendu est issu de ce que j'ai fait...

[OnMove500HR.cc.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/6167863/OnMove500HR.cc.txt)
[OnMove500HR.h.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/6167864/OnMove500HR.h.txt)
[ACT_0060.OMD.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/6167868/ACT_0060.OMD.txt)
[ACT_0060.OMH.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/6167869/ACT_0060.OMH.txt)
[case14.zip.txt](https://github.com/ColinPitrat/kalenji-gps-watch-reader/files/6167876/case14.zip.txt)

Je prévois de continuer à travailler dessus pour tirer le maximum de ma montre... Sans doute regarder du côté de l'extension GPXdata, pour enregistrer les valeurs énergie et vitesse qui ne sont pas utilsés (cela pourra servir sur tous les modèles...)
Le OnMove 500 enregistre une valeur d'énergie à chaque point, mais cela n'est pas prévu dans les propriétés de l'objet point...

BBllrd, 2022-07-17:
Bonjour,
J'ai repris mon travail qui ne fonctionnait pas correctement... 
Cette fois ci j'ai travaillé à partir d'un fork... Sans doute plus facile à reprendre... 
https://github.com/BBllrd/kalenji-gps-watch-reader-bbl
Encore merci !

ColinPitrat, 2023-05-10:
Désolé pour le long silence. Pourriez vous m'envoyer une Pull Request?